Azure/Industrial-IoT #142: OPC Publisher: Self-Signed Certificate Creation Issue

YashC, 2019-10-29:
OPC Publisher creating a self signed application certificate on deployment, this was working until recently but is failing now.  I couldn't replicate on another machine. 

Running on window edge.

PS C:\Windows\system32> iotedge check --verbose
Configuration checks
--------------------
√ config.yaml is well-formed - OK
√ config.yaml has well-formed connection string - OK
√ container engine is installed and functional - OK
√ Windows host version is supported - OK
√ config.yaml has correct hostname - OK
√ config.yaml has correct URIs for daemon mgmt endpoint - OK
√ latest security daemon - OK
√ host time is close to real time - OK
√ container time is close to host time - OK
√ DNS server - OK
‼ production readiness: certificates - Warning
    Device is using self-signed, automatically generated certs.
    Please see https://aka.ms/iotedge-prod-checklist-certs for best practices.
√ production readiness: certificates expiry - OK
√ production readiness: container engine - OK
√ production readiness: logs policy - OK

Connectivity checks
-------------------
√ host can connect to and perform TLS handshake with IoT Hub AMQP port - OK
√ host can connect to and perform TLS handshake with IoT Hub HTTPS / WebSockets port - OK
√ host can connect to and perform TLS handshake with IoT Hub MQTT port - OK
√ container on the IoT Edge module network can connect to IoT Hub AMQP port - OK
√ container on the IoT Edge module network can connect to IoT Hub HTTPS / WebSockets port - OK
√ container on the IoT Edge module network can connect to IoT Hub MQTT port - OK
√ Edge Hub can bind to ports on host - OK

20 check(s) succeeded.
1 check(s) raised warnings.

  
**Container Create Options:** 
{
  "Hostname": "publisher",
  "Cmd": [
    "publisher",
    "--pf=\\publisher\\pn.json",
    "--di=60",
    "--to",
    "--aa",
    "--si=0",
    "--ms=512",
    "--mq=32768",
    "--ss=0x80000000, BadNoCommunication, BadWaitingForInitialData",
    "--appcertstoretype=X509Store",
    "--oi=120000",
    "--op=240000",
    "--sf=True",
    "--ll=verbose"
  ],
  "ExposedPorts": {
    "62222/tcp": {}
  },
  "NetworkingConfig": {
    "EndpointsConfig": {
      "host": {
        "IPAMConfig": {
          "IPv4Address": "xx.xx.xx.xx"
        }
      }
    }
  },
  "HostConfig": {
[... 70 lines cut ...]
2019-10-29T20:09:14.692548800Z    at CertificateFactory.CreateCertificateWithPrivateKey(X509Certificate certificate, String friendlyName, AsymmetricKeyParameter privateKey, SecureRandom random)
2019-10-29T20:09:14.692548800Z    at CertificateFactory.CreateCertificate(String storeType, String storePath, String password, String applicationUri, String applicationName, String subjectName, IList1 domainNames, UInt16 keySize, DateTime startTime, UInt16 lifetimeInMonths, UInt16 hashSizeInBits, Boolean isCA, X509Certificate2 issuerCAKeyCert, Byte[] publicKey, Int32 pathLengthConstraint)
2019-10-29T20:09:14.692548800Z    at OpcPublisher.OpcApplicationConfiguration.InitApplicationSecurityAsync() in D:\a\1\s\modules\opc-publisher\opcpublisher\OpcApplicationConfigurationSecurity.cs:line 197
2019-10-29T20:09:14.692548800Z    at OpcPublisher.OpcApplicationConfiguration.ConfigureAsync() in D:\a\1\s\modules\opc-publisher\opcpublisher\OpcApplicationConfiguration.cs:line 172
2019-10-29T20:09:14.692548800Z    at OpcPublisher.Program.MainAsync(String[] args) in D:\a\1\s\modules\opc-publisher\opcpublisher\Program.cs:line 697
2019-10-29T20:09:14.692548800Z System.NotSupportedException: Creating X509Certificate from PKCS #12 store failed ---> System.Security.Cryptography.CryptographicException: The public/private key pair in the certficates do not match.
2019-10-29T20:09:14.692548800Z    at CertificateFactory.VerifyRSAKeyPair(X509Certificate2 certWithPublicKey, X509Certificate2 certWithPrivateKey, Boolean throwOnError)
2019-10-29T20:09:14.692548800Z    at CertificateFactory.CreateCertificateFromPKCS12(Byte[] rawData, String password)
2019-10-29T20:09:14.692548800Z    --- End of inner exception stack trace ---
2019-10-29T20:09:14.692548800Z    at CertificateFactory.CreateCertificateFromPKCS12(Byte[] rawData, String password)
2019-10-29T20:09:14.692548800Z    at CertificateFactory.CreateCertificateWithPrivateKey(X509Certificate certificate, String friendlyName, AsymmetricKeyParameter privateKey, SecureRandom random)
2019-10-29T20:09:14.692548800Z    at CertificateFactory.CreateCertificate(String storeType, String storePath, String password, String applicationUri, String applicationName, String subjectName, IList1 domainNames, UInt16 keySize, DateTime startTime, UInt16 lifetimeInMonths, UInt16 hashSizeInBits, Boolean isCA, X509Certificate2 issuerCAKeyCert, Byte[] publicKey, Int32 pathLengthConstraint)
2019-10-29T20:09:14.692548800Z    at OpcPublisher.OpcApplicationConfiguration.InitApplicationSecurityAsync() in D:\a\1\s\modules\opc-publisher\opcpublisher\OpcApplicationConfigurationSecurity.cs:line 197
2019-10-29T20:09:14.692548800Z    at OpcPublisher.OpcApplicationConfiguration.ConfigureAsync() in D:\a\1\s\modules\opc-publisher\opcpublisher\OpcApplicationConfiguration.cs:line 172
2019-10-29T20:09:14.692548800Z    at OpcPublisher.Program.MainAsync(String[] args) in D:\a\1\s\modules\opc-publisher\opcpublisher\Program.cs:line 697
2019-10-29T20:09:14.694443700Z [04:09:14 FTL]    at CertificateFactory.VerifyRSAKeyPair(X509Certificate2 certWithPublicKey, X509Certificate2 certWithPrivateKey, Boolean throwOnError)
2019-10-29T20:09:14.694443700Z    at CertificateFactory.CreateCertificateFromPKCS12(Byte[] rawData, String password)
2019-10-29T20:09:14.694443700Z System.Security.Cryptography.CryptographicException: The public/private key pair in the certficates do not match.
2019-10-29T20:09:14.694443700Z    at CertificateFactory.VerifyRSAKeyPair(X509Certificate2 certWithPublicKey, X509Certificate2 certWithPrivateKey, Boolean throwOnError)
2019-10-29T20:09:14.694443700Z    at CertificateFactory.CreateCertificateFromPKCS12(Byte[] rawData, String password)
2019-10-29T20:09:14.694443700Z [04:09:14 FTL] Publisher exiting...`

Not sure what i'm missing, Any help? 

Thanks

markusstuhler, 2019-11-05:
Hi @YashC ,

could you try the solution described here to see if it fixes the issue: https://github.com/OPCFoundation/UA-.NETStandard/issues/550

Thanks!
Markus

markusstuhler, 2019-11-18:
Closing due to inactivity. Please reopen if you have further questions.

Thanks!

closed 2019-11-18 by markusstuhler

FMauroy, 2020-04-14:
Hi all,

we are having the exact same issue here. We never had that problem elsewhere, so we are a bit puzzled. It has happened twice in a week now, only at one customer. We run Windows 10 IoT Enterprise 17763 LTSC.

The weird part is that it worked fine for some time, and today, after restarting the machine, OPC Publisher failed to start. This is the complete log:

```
[15:41:34 INF] OPC Publisher V2.4.1+1 starting up...
[15:41:34 DBG] Informational version: V2.4.1+1.Branch.master.Sha.d5a561db5b6d4a8590d1783e939f277a3de02b84
[15:41:34 INF] OPC UA monitored item notifications with one of the following 2 status codes will not generate telemetry events:
[15:41:34 INF] StatusCode: BadNoCommunication (dec: 2150694912, hex: 80310000)
[15:41:34 INF] StatusCode: BadWaitingForInitialData (dec: 2150760448, hex: 80320000)
[15:41:34 INF] opcstacktracemask set to: 0x7FFFFFFF
[15:41:34 INF] OperationTimeout set to 120000
[15:41:34 INF] OPC UA server base address: opc.tcp://publisher:62222/UA/Publisher
[15:41:34 INF] Security policy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256 with mode SignAndEncrypt added
[15:41:34 INF] Trusted Issuer store type is: Directory
[15:41:34 INF] Trusted Issuer Certificate store path is: pki/issuer
[15:41:34 INF] Trusted Peer Certificate store type is: Directory
[15:41:34 INF] Trusted Peer Certificate store path is: pki/trusted
[15:41:34 INF] Rejected certificate store type is: Directory
[15:41:34 INF] Rejected Certificate store path is: pki/rejected
[15:41:34 INF] Rejection of SHA1 signed certificates is disabled
[15:41:34 INF] Minimum certificate key size set to 1024
[15:41:34 INF] Application Certificate store type is: Directory
[15:41:34 INF] Application Certificate store path is: pki/own
[15:41:34 INF] Application Certificate subject name is: publisher
[15:41:34 WRN] WARNING: Automatically accepting certificates. This is a security risk.
[15:41:34 INF] No existing Application certificate found. Create a self-signed Application certificate valid from yesterday for 12 months,
[15:41:34 INF] with a 2048 bit key and 256 bit hash.
[15:41:35 FTL]    at CertificateFactory.CreateCertificateFromPKCS12(Byte[] rawData, String password)
   at CertificateFactory.CreateCertificateWithPrivateKey(X509Certificate certificate, String friendlyName, AsymmetricKeyParameter privateKey, SecureRandom random)
   at CertificateFactory.CreateCertificate(String storeType, String storePath, String password, String applicationUri, String applicationName, String subjectName, IList`1 domainNames, UInt16 keySize, DateTime startTime, UInt16 lifetimeInMonths, UInt16 hashSizeInBits, Boolean isCA, X509Certificate2 issuerCAKeyCert, Byte[] publicKey, Int32 pathLengthConstraint)
   at OpcPublisher.OpcApplicationConfiguration.InitApplicationSecurityAsync() in C:\app\opcpublisher\OpcApplicationConfigurationSecurity.cs:line 197
   at OpcPublisher.OpcApplicationConfiguration.ConfigureAsync() in C:\app\opcpublisher\OpcApplicationConfiguration.cs:line 173
   at OpcPublisher.Program.MainAsync(String[] args) in C:\app\opcpublisher\Program.cs:line 695
System.NotSupportedException: Creating X509Certificate from PKCS #12 store failed ---> System.Security.Cryptography.CryptographicException: The public/private key pair in the certficates do not match.
   at CertificateFactory.VerifyRSAKeyPair(X509Certificate2 certWithPublicKey, X509Certificate2 certWithPrivateKey, Boolean throwOnError)
   at CertificateFactory.CreateCertificateFromPKCS12(Byte[] rawData, String password)
   --- End of inner exception stack trace ---
   at CertificateFactory.CreateCertificateFromPKCS12(Byte[] rawData, String password)
   at CertificateFactory.CreateCertificateWithPrivateKey(X509Certificate certificate, String friendlyName, AsymmetricKeyParameter privateKey, SecureRandom random)
   at CertificateFactory.CreateCertificate(String storeType, String storePath, String password, String applicationUri, String applicationName, String subjectName, IList`1 domainNames, UInt16 keySize, DateTime startTime, UInt16 lifetimeInMonths, UInt16 hashSizeInBits, Boolean isCA, X509Certificate2 issuerCAKeyCert, Byte[] publicKey, Int32 pathLengthConstraint)
   at OpcPublisher.OpcApplicationConfiguration.InitApplicationSecurityAsync() in C:\app\opcpublisher\OpcApplicationConfigurationSecurity.cs:line 197
   at OpcPublisher.OpcApplicationConfiguration.ConfigureAsync() in C:\app\opcpublisher\OpcApplicationConfiguration.cs:line 173
   at OpcPublisher.Program.MainAsync(String[] args) in C:\app\opcpublisher\Program.cs:line 695
[15:41:35 FTL]    at CertificateFactory.VerifyRSAKeyPair(X509Certificate2 certWithPublicKey, X509Certificate2 certWithPrivateKey, Boolean throwOnError)
   at CertificateFactory.CreateCertificateFromPKCS12(Byte[] rawData, String password)
System.Security.Cryptography.CryptographicException: The public/private key pair in the certficates do not match.
   at CertificateFactory.VerifyRSAKeyPair(X509Certificate2 certWithPublicKey, X509Certificate2 certWithPrivateKey, Boolean throwOnError)
   at CertificateFactory.CreateCertificateFromPKCS12(Byte[] rawData, String password)
[15:41:35 FTL] Publisher exiting...
```

I "solved" it by changing the certificates storage location, from Directory to X509Store. The previous time, I tried it and it failed, and only worked after I reverted this change back to Directory, and it kept running for several days.

So, we are a bit lost on how to determine the cause of this issue and more importantly on what to do to fix it for good.

Does anyone have any idea?

Thanks a lot!
fred

bensalem22, 2023-03-17:
I also have the same issue:
[2023-03-17 14:15:38.950 INF OpcUa] Checking application instance certificate.
[2023-03-17 14:15:39.037 INF OpcUa] Imported the PFX private key for [A9106BB591BB06A0A0ACC0CEEE6770340684F3F5].
[2023-03-17 14:15:39.039 INF OpcUa] Check certificate: [CN=Microsoft.Azure.IIoT, C=DE, S=Bav, O=Microsoft, DC=mada030014.mad03.shopfloor.net] [A9106BB591BB06A0A0ACC0CEEE6770340684F3F5]
[2023-03-17 14:15:39.041 INF OpcUa] Check application instance certificate. [CN=Microsoft.Azure.IIoT, C=DE, S=Bav, O=Microsoft, DC=mada030014.mad03.shopfloor.net] [A9106BB591BB06A0A0ACC0CEEE6770340684F3F5]
[2023-03-17 14:15:39.121 INF OpcUa] Using the ApplicationUri: urn:mada030014.mad03.shopfloor.net:Microsoft.Azure.IIoT:microsoft:
[2023-03-17 14:15:39.139 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Application own certificate store contains 1 certs.
[2023-03-17 14:15:39.139 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] 01: Subject 'CN=Microsoft.Azure.IIoT, C=DE, S=Bav, O=Microsoft, DC=mada030014.mad03.shopfloor.net' (thumbprint: A9106BB591BB06A0A0ACC0CEEE6770340684F3F5)
[2023-03-17 14:15:39.140 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Trusted issuer store contains 0 certs.
[2023-03-17 14:15:39.141 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Trusted issuer store has 0 CRLs.
[2023-03-17 14:15:39.142 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Trusted peer store contains 1 certs.
[2023-03-17 14:15:39.142 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] 01: Subject 'CN=Microsoft.Azure.IIoT, C=DE, S=Bav, O=Microsoft, DC=mada030014.mad03.shopfloor.net' (thumbprint: A9106BB591BB06A0A0ACC0CEEE6770340684F3F5)
[2023-03-17 14:15:39.142 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Trusted peer store has 0 CRLs.
[2023-03-17 14:15:39.142 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Rejected certificate store contains 0 certs.

any recommendations?