Azure/Industrial-IoT #143: Messages lost due to message queue above capacity

scottmur, 2019-10-30:
**Describe the bug**
The internal monitored item message queue is above its capacity of 250000. 

**To Reproduce**
Steps to reproduce the behavior:
Intermittent issue.  Result is visible via TSI client showing messages lost. refer to attached image

**Expected behavior**
Messages are not lost and are routed successfully upstream

**Screenshots**

[image]

**IOT Edge Server (please complete the following information):**
 - OS: Ubuntu 18.04.2 LTS (GNU/Linux 4.15.0-55-generic x86_64)
edgeHub                running          Up 2 weeks       mcr.microsoft.com/azureiotedge-hub:1.0.8.2
opcpublisher           running          Up 2 weeks       mcr.microsoft.com/iotedge/opc-publisher:2.5.1
edgeAgent              running          Up 2 weeks       mcr.microsoft.com/azureiotedge-agent:1.0.8.2

**Additional context**

Once receive this error do not seem to recover.  Restarting application resolves issue for period of time
The internal monitored item message queue is above its capacity of 250000. We have already lost 114980000 monitored item notifications:(

markusstuhler, 2019-11-06:
Hi @scottmur 

could you share the pn.json file? If the queue is running full, then more messages are issued by the OPC UA server then can be sent to IoT Hub. Could you try to increase the sampling/publishing intervals or reduce the number of monitored nodes to check if this is the issue?

Thanks!
Markus

scottmur, 2019-11-07:
Markus

Currently we are consuming twice as fast so queue size should not be an issue

Each opc node is similar config to this:
                "OpcSamplingInterval": 30000,
                "OpcPublishingInterval": 15000,
                "HeartbeatInterval": 360

Don’t believe what we are seeing is a queue issue

Cheers

Murray

Murray Scott
Senior Services Engineer, AOT, Global Assets and Technology

[cid:image002.png@01D59575.5EFE8B30]
Fonterra Co-operative Group Limited
murray.scott@fonterra.com<mailto:murray.scott@fonterra.com>
direct +64 7 871 2459
mobile + 64 274 966 622
PO Box 10397, Hamilton 3241
1344 Te Rapa Road, Horotiu, Hamilton 3288, New Zealand

[Social Media Icons-07]<http://www.fonterra.com/> [Social Media Icons-10] <http://www.twitter.com/fonterra>  [Social Media Icons-11] <http://www.facebook.com/fonterra>  [Social Media Icons-12] <https://www.instagram.com/fonterra.nz/>  [Social Media Icons-08] <http://www.youtube.com/fonterraltd>  [Social Media Icons-09] <http://www.linkedin.com/company/fonterra>

barnstee, 2020-09-17:
Consumption in OPC Publisher is not the issue, but network latency to IoT Hub. If OPC Publisher drops messages it is because it can't send them fast enough to IoT Hub. This is by design and backpressure handling. Please reduce the number of messages you are sending or setup a second OPC Publisher in parallel. I'm closing for now. If you still experience issues, please reopen.

closed 2020-09-17 by barnstee