Azure/Industrial-IoT #144: Missing info on JSON payload

lucarv, 2019-11-03:
https://github.com/Azure/Industrial-IoT/blob/master/docs/modules/publisher.md in the telemetry format section explains that "ApplicationUri": "urn:myopcserver" is sent as part of the default payload if no special config file is denied with --tc flag.
This is what I am getting:
 ```
[IoTHubMonitor] [11:00:42 AM] Message received from [edgeVM4opcua/OPCPublisher]:
{
  "NodeId": "ns=1;s=freemem",
  "DisplayName": "Available RAM",
  "Value": {
    "Value": 6306189312,
    "SourceTimestamp": "2019-11-03T10:00:39.5083379Z"
  }
}```

markusstuhler, 2019-11-06:
Hi @lucarv,

that's correct, the ApplicationUri is set as being sent with default settings. Can you share the command line parameters and the version of the publisher that you are running?

Thanks!
Markus

lucarv, 2019-11-06:
I am using the default parameters that came with the version in the marketolace

markusstuhler, 2020-02-20:
I couldn't reproduce this on my end. Please try if the issue persists with the latest version and reopen if so.

Thanks!
Markus

closed 2020-02-20 by markusstuhler