Azure/Industrial-IoT #147: Forked opcPublisher TLS Issue

TMG-DG, 2019-11-07:
We forked the latest opcPublisher release, and added some basic code to pull the Server Time from Kepware, since doing so we experience issues with the Module handshake to IoT Hub

[15:30:48 ERR] Failure initializing hub communication processing.
System.Security.Authentication.AuthenticationException: TLS authentication error. ---> System.Security.Authentication.AuthenticationException: The remote certificate is invalid according to the validation procedure.
   at System.Net.Security.SslState.StartSendAuthResetSignal(ProtocolToken message, AsyncProtocolRequest asyncRequest, ExceptionDispatchInfo exception)
   at System.Net.Security.SslState.CheckCompletionBeforeNextReceive(ProtocolToken message, AsyncProtocolRequest asyncRequest)
   at System.Net.Security.SslState.StartSendBlob(Byte[] incoming, Int32 count, AsyncProtocolRequest asyncRequest)
   at System.Net.Security.SslState.ProcessReceivedBlob(Byte[] buffer, Int32 count, AsyncProtocolRequest asyncRequest)
   at System.Net.Security.SslState.StartReadFrame(Byte[] buffer, Int32 readBytes, AsyncProtocolRequest asyncRequest)
   at System.Net.Security.SslState.StartReceiveBlob(Byte[] buffer, AsyncProtocolRequest asyncRequest)
   at System.Net.Security.SslState.CheckCompletionBeforeNextReceive(ProtocolToken message, AsyncProtocolRequest asyncRequest)
   at System.Net.Security.SslState.StartSendBlob(Byte[] incoming, Int32 count, AsyncProtocolRequest asyncRequest)
   at System.Net.Security.SslState.ProcessReceivedBlob(Byte[] buffer, Int32 count, AsyncProtocolRequest asyncRequest)
   at System.Net.Security.SslState.StartReadFrame(Byte[] buffer, Int32 readBytes, AsyncProtocolRequest asyncRequest)
   at System.Net.Security.SslState.StartReceiveBlob(Byte[] buffer, AsyncProtocolRequest asyncRequest)
   at System.Net.Security.SslState.CheckCompletionBeforeNextReceive(ProtocolToken message, AsyncProtocolRequest asyncRequest)
   at System.Net.Security.SslState.StartSendBlob(Byte[] incoming, Int32 count, AsyncProtocolRequest asyncRequest)
   at System.Net.Security.SslState.ProcessReceivedBlob(Byte[] buffer, Int32 count, AsyncProtocolRequest asyncRequest)
   at System.Net.Security.SslState.StartReadFrame(Byte[] buffer, Int32 readBytes, AsyncProtocolRequest asyncRequest)
   at System.Net.Security.SslState.PartialFrameCallback(AsyncProtocolRequest asyncRequest)
--- End of stack trace from previous location where exception was thrown ---

At first we thought this was caused by our Production certificates which we had started to recently roll out, we had observed that the issue did not seem to occur when Dev certs where in use. Until today that is...

The site that the logs above are from, is running DEV certs, and the custom opcPublisher module has been running for quite sometime. Then we noticed the stream go down, and we see the same issue.

Redeploying the MSFT Version 2.4.1 of the opcPublisher allows it to work, but we can't pinpoint why the TLS issue is occuring.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
opcPublisher operates as normal without TLS Issues

**Desktop (please complete the following information):**
 - OS: Linux - Ubuntu 18.04
- opcPublisher
 - Version: 2.4.1

markusstuhler, 2019-11-18:
Hi @TMG-DG,

Could you double check if the certificate, that is used is valid? The message is coming from the framework itself.
Thanks!

TMG-DG, 2019-11-18:
Hi @markusstuhler 

Yes, certificate is valid, all other modules that we have deployed and communicate with IoT Hub work without any issues. In addition, the MSFT Versions of Publisher seem to operate correctly with the Production Certs.

Iotedge check does not report any issues with the certificates either.

Cert:

Signature Algorithm: sha256WithRSAEncryption
        Issuer: CN = The Marsden Group Intermediate
        Validity
            Not Before: Nov 18 13:59:52 2019 GMT
            Not After : Jan 26 14:00:22 2022 GMT
        Subject: CN = manual-tmg-dell-03-01

Certification Verification works as expected

markusstuhler, 2019-11-19:
Hi @TMG-DG, 

please try the latest version of the .NET Core SDK and Azure Devices SDK. Since the issue does not occur in the official build, this must be related to the fork. As such, I'm closing this issue.

Thanks!

closed 2019-11-19 by markusstuhler