Azure/Industrial-IoT #148: OPC Publisher won't send messages to IoT Hub (standalone edge module)

awoSYS, 2019-11-08:
Hi! I'm facing a problem using OPC Publisher as standalone edge module: it doesn't seem to send any messages to IoTHub. We're just trying to connect to an unsecured OPC UA server and stream data from a single node to IoT Hub. Any help appreciated!

**Describe the bug**
I am running OPC Publisher as standalone Azure IoT Edge module. Even though the module's periodic logs report an increasing number of "messages sent to IoTHub", neither edgeHub module nor IoT Hub seem to receive any messages. 

**To Reproduce**
1. Deploy edge module via deployment manifest: (via Visual Studio Code)
```json
...
"modules": {
          "opcpublisher": {
            "version": "2.0",
            "type": "docker",
            "status": "running",
            "restartPolicy": "always",
            "settings": {
              "image": "mcr.microsoft.com/iotedge/opc-publisher:latest",
              "createOptions": "{\"Hostname\": \"publisher\", \"Cmd\": [ \"publisher\", \"--pf=./pn.json\", \"--di=60\", \"--to\", \"--aa\"]}"
            }
          }
        }
      }
    },
    "$edgeHub": {
      "properties.desired": {
        "schemaVersion": "1.0",
        "routes": {
          "opcpublisherToIoTHub": "FROM /messages/modules/opcpublisher/outputs/* INTO $upstream"
        },
...
```
2. Configure OPC UA node to be published by calling direct method `PublishNodes` with payload
```json
{
   "EndpointUrl": "opc.tcp://opc.server.url:53530/OPCUA/SimulationServer",
   "UseSecurity": false,
   "OpcNodes": [
      {
        "Id": "ns=5;s=Random1",
        "OpcSamplingInterval": 2000,
        "OpcPublishingInterval": 5000,
        "HeartbeatInterval": 3600,
        "SkipFirst":  true
      }
   ]
}
  ```

**Expected behavior**
- edgeHub module logs should show something like `opcpublisherToIoTHub: FROM /messages/modules/opcpublisher/outputs/* INTO $upstream` every second or so
- IoTHub's default endpoint monitoring (via Visual Studio Code) should print some json messages every second

**Logs**
- Publisher logs:
```
[14:16:56 INF] Publisher is running. Press CTRL-C to quit.
[14:17:47 INF] HandlePublishNodesMethodAsync: No matching session found for endpoint 'opc.tcp://<url>:53530/OPCUA/SimulationServer'. Requested to create a new one.
[14:17:47 INF] AddNodeForMonitoringAsync: No matching subscription with publishing interval of 5000 found.
[14:17:47 INF] Create a new subscription with a publishing interval of 5000.
[14:17:47 INF] HandlePublishNodesMethodAsync: completed with result OK
[14:17:47 INF] Connect and monitor session and nodes on endpoint 'opc.tcp://<url>:53530/OPCUA/SimulationServer'.
[14:17:48 INF] Create unsecured session for endpoint URI 'opc.tcp://<url>:53530/OPCUA/SimulationServer' with timeout of 10000 ms.
[14:17:48 INF] Certificate 'DC=<url>, O=Prosys OPC, CN=SimulationServer' will be trusted, because of corresponding command line option.
[14:17:48 INF] Session successfully created with Id ns=1;g=e7a2c2be-3a9c-47af-a22d-9995eea49089.
[14:17:48 INF] The session to endpoint 'opc.tcp://<url>:53530/OPCUA/SimulationServer' has 7 entries in its namespace array:
[14:17:48 INF] Namespace index 0: http://opcfoundation.org/UA/
[14:17:48 INF] Namespace index 1: urn:<url>:OPCUA:SimulationServer
[14:17:48 INF] Namespace index 2: http://www.prosysopc.com/OPCUA/SampleAddressSpace
[14:17:48 INF] Namespace index 3: http://www.prosysopc.com/OPCUA/ComplianceNodes
[14:17:48 INF] Namespace index 4: http://www.prosysopc.com/OPCUA/ComplianceNonUaNodes
[... 71 lines cut ...]
<6> 2019-11-08 14:16:51.232 +00:00 [INF] - Closing link MethodSending for opc-ua-poc-vm/opcpublisher
<6> 2019-11-08 14:16:51.233 +00:00 [INF] - Closing link MethodReceiving for opc-ua-poc-vm/opcpublisher
<6> 2019-11-08 14:16:51.233 +00:00 [INF] - Setting proxy inactive for opc-ua-poc-vm/opcpublisher.
<6> 2019-11-08 14:16:51.234 +00:00 [INF] - Device connection removed for device opc-ua-poc-vm/opcpublisher
<6> 2019-11-08 14:16:51.260 +00:00 [INF] - Remove device connection for device opc-ua-poc-vm/opcpublisher
<6> 2019-11-08 14:16:51.379 +00:00 [INF] - Updated reported properties for opc-ua-poc-vm/$edgeHub
<6> 2019-11-08 14:16:56.653 +00:00 [INF] - New token received on the Cbs link
<6> 2019-11-08 14:16:56.704 +00:00 [INF] - Client opc-ua-poc-vm/opcpublisher in device scope authenticated locally.
<6> 2019-11-08 14:16:56.705 +00:00 [INF] - New device connection for device opc-ua-poc-vm/opcpublisher
<6> 2019-11-08 14:16:56.708 +00:00 [INF] - Bind device proxy for device opc-ua-poc-vm/opcpublisher
<6> 2019-11-08 14:16:56.708 +00:00 [INF] - Initialized device listener in the AMQP protocol head for opc-ua-poc-vm/opcpublisher
<6> 2019-11-08 14:16:56.708 +00:00 [INF] - Opened link Events for opc-ua-poc-vm/opcpublisher
<6> 2019-11-08 14:16:56.746 +00:00 [INF] - Opened link MethodSending for opc-ua-poc-vm/opcpublisher
<6> 2019-11-08 14:16:56.751 +00:00 [INF] - Opened link MethodReceiving for opc-ua-poc-vm/opcpublisher
<6> 2019-11-08 14:16:56.847 +00:00 [INF] - Updated reported properties for opc-ua-poc-vm/$edgeHub
<6> 2019-11-08 14:20:02.001 +00:00 [INF] - Reauthenticating connected clients
<6> 2019-11-08 14:25:02.008 +00:00 [INF] - Reauthenticating connected clients
```

**Desktop (please complete the following information):**
 ```
sudo iotedge --version
iotedge 1.0.8 (208b2204fd30e856d00b280112422130c104b9f0)
```
- OS: Ubuntu 18.04

awoSYS, 2019-11-11:
Some updates:

- I set the logging level environment variable to `verbose` and even saw the following log line:
```
[15:06:44 DBG] Sending 453 bytes to hub.
```
- Tried to catch messages arriving in IoTHub with an additional StreamAnalytics Job and manually sampled data from it - with empty result set.
- Also tried to start the container with an additional connection string argument to bypass edge module `edgeHub` and directly send messages to IoTHub - with no success.

Any help appreciated. ;)

holgerbrandl, 2019-11-12:
Similar problem here: Data is correctly read out from the connected OPC-UA server, is reported as being sent (when using verbose log level), but the event stream in the IoT-Hub stays empty. Also bypassing the edgehub by providing a connection string for iothub gives the same incorrect result (no events).

markusstuhler, 2019-11-19:
Hi @awoSYS and @holgerbrandl 

We could not reproduce this on our side. However, could you try to use a new deployed "plain" IoT Hub instance to send the data via connection string to? The source of the issue could be when IoT Hub limits are reached and messages are throttled.

Thanks!

awoSYS, 2019-11-21:
Hi markusstuhler! 
Thanks for your reply! The IoT Hub I used was freshly created and used only together with this single edge device. As I pointed out, I already tried the alternative connection string method for this purpose (and created a new IoT device entity in IoT Hub to get the connection string). No success.
Anyway, even if IoT Hub would be causing the problem, wouldn't I still see the messages being transferred from OPC Publisher to Edge Hub module (in Edge Hub logs) before sending to IoT Hub fails?

marcschier, 2020-03-27:
Can you try: "/messages/modules/opcpublisher/*".  We do not use outputs and this is likely the issue.

closed 2020-03-27 by marcschier