Azure/Industrial-IoT #149: OPCPubslisher can not read config file

Bananenbrot1, 2019-11-14:
Hi,

 i tried to implement the OPCPublisher module on our edge device. I installed the module from the azure marketplace and deployed it with our IoT edge hub. The module starts like expected but when i try to add a config file, following error message appears.

```
Attemtping to load node configuration from: /appdata/pn.json
[16:50:11 INF] Loaded 1 config file entry/entries.
[16:50:11 FTL] Loading of the node configuration file failed. Does the file exist and has correct syntax? Exiting...
System.NullReferenceException: Object reference not set to an instance of an object.
   at OpcPublisher.PublisherNodeConfiguration.ReadConfigAsync() in D:\a\1\s\modules\opc-publisher\opcpublisher\PublisherNodeConfiguration.cs:line 361
[16:50:11 ERR] Error while reading the node configuration file '/appdata/pn.json'
[16:50:11 FTL]    at OpcPublisher.PublisherNodeConfiguration..ctor() in D:\a\1\s\modules\opc-publisher\opcpublisher\PublisherNodeConfiguration.cs:line 259
   at OpcPublisher.PublisherNodeConfiguration.get_Instance() in D:\a\1\s\modules\opc-publisher\opcpublisher\PublisherNodeConfiguration.cs:line 229
   at OpcPublisher.Program.MainAsync(String[] args) in D:\a\1\s\modules\opc-publisher\opcpublisher\Program.cs:line 740
System.Exception: Error while reading the node configuration file '/appdata/pn.json'
   at OpcPublisher.PublisherNodeConfiguration..ctor() in D:\a\1\s\modules\opc-publisher\opcpublisher\PublisherNodeConfiguration.cs:line 259
   at OpcPublisher.PublisherNodeConfiguration.get_Instance() in D:\a\1\s\modules\opc-publisher\opcpublisher\PublisherNodeConfiguration.cs:line 229
   at OpcPublisher.Program.MainAsync(String[] args) in D:\a\1\s\modules\opc-publisher\opcpublisher\Program.cs:line 740
[16:50:11 FTL] Publisher exiting... 
```

my config file looks like this: (the ip adress and port are normal IP adresses but i blanked them out for security reasons.)

```json
[
        {
                "EndpointUrl":"opc.tcp://ip-adress:port",
                "UseSecurity":false
        }
]
```

The config file is loaded correctly into the docker container and i gave the file 777 on my host system.

When i try to load the file without payload there is no error. 

The edge device is running an ubuntu 16.04

marcschier, 2019-11-15:
Hi @Bananenbrot1 - 

This looks like as if this happened when there are no nodes specified that should be monitored.   I would say this is absolutely a bug, as it should not crash.   If you intend to monitor nodes using the configuration file then please add the nodes you would like monitored.  If you intend to use the device methods API (soon to be deprecated) then just keep the empty array and remove the existing object.

markusstuhler, 2019-11-18:
Hi @Bananenbrot1, 

we will include this in our backlog to improve the error message. However, the solution is like Marc described: you can specify monitored nodes along with the endpoint or no endpoint at all.

Closing this issue. Please reopen if you have further questions.

Thanks!

closed 2019-11-18 by markusstuhler