Azure/Industrial-IoT #153: can't start publisher module

lucarv, 2019-11-16:
**Describe the bug**
Mounting a volume with container create options in IoT Edge will stop edge to be able to start it.

**To Reproduce**
Steps to reproduce the behavior:
1. Install the publisher:latest anyway you like (i do it from market place
2. Verify that the publisher starts and is healthy. Also verify that it does not read the config file as no volumes are mounted.
```
[07:04:23 INF] The name of the configuration file for published nodes is: /app/publishednodes.json
[07:04:23 INF] The node configuration file '/app/publishednodes.json' does not exist. Continue and wait for remote configuration requests.
```
3. Bind the volume in the Deployment manifest according to what is documented in https://docs.microsoft.com/en-us/azure/iot-accelerators/howto-opc-publisher-run#make-the-configuration-files-accessible-on-the-host. Edit so it have the mount point as /app as this is the default location.  Also don't use the --pf switch and let the default name (publishednodes.json) be used
```
{
    "Hostname": "publisher",
    "Cmd": [
        "--aa"
    ],
    "HostConfig": {
        "Binds": [
            "/home/iotedge/appdata:/app"
        ]
    }
}
4. Publish the manifest to the Edge device.
5. See error from edgeAgent
```
<6> 2019-11-16 07:03:19.234 +00:00 [INF] - Plan execution ended for deployment 245
<3> 2019-11-16 07:03:19.234 +00:00 [ERR] - Edge agent plan execution failed.
System.AggregateException: One or more errors occurred. (Error calling start module OPCPublisher: Could not start module OPCPublisher
	caused by: Could not start module OPCPublisher
	caused by: OCI runtime create failed: container_linux.go:345: starting container process caused "exec: \"./opcpublisher\": stat ./opcpublisher: no such file or directory": unknown) ---> Microsoft.Azure.Devices.Edge.Agent.Edgelet.EdgeletCommunicationException: Error calling start module OPCPublisher: Could not start module OPCPublisher
	caused by: Could not start module OPCPublisher
```
Note that there is more on the log trace but this should be enough, otherwise it is easy to reproduce.

**Expected behavior**
edgeAgent should start the publisher module..

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Ubuntu 18.04
 - Browser N/A

**Additional context**
Using the publisher:2.3 one is able to mount the volume and run the module.

markusstuhler, 2019-11-19:
This is a bug in our updated build scripts, we will change this in 2.5.2 hotfix.

dominicbetts, 2019-11-21:
@markusstuhler  - do we need to make and changes to the docs? There is a [docs issue](https://github.com/MicrosoftDocs/azure-docs/issues/42909) open that refers to this.

markusstuhler, 2019-12-03:
Hi,

This has been fixed in 2.5.2.

Please reopen if you have further issues.

Thanks!
Markus

closed 2019-12-03 by markusstuhler