Azure/Industrial-IoT #161: New Deployment Procedure - Admin consent required

mma-mm, 2019-11-25:
Hi,

i tried to deploy the Industrial-IoT-Solution using the new Deployment program.
I used it without any parameters, so it asked me for information. After providing tenant id and logging in, i was redirected to admin consent, which i cannot give, as i'm not the admin of the AAD.

I'm Owner of the subscription i want to use have sufficient rights to deploy using the old deployment procedure. (Creating, changing app registration etc.)

How is determined, whether admin consent is required? What is the admin consent required for?

Also i couldn't find any documentation of the new deployment procedure. Where do i have to look?

Thank you for your help

Matthias

karok2m, 2019-11-29:
Hi Matthias,

We are using Microsoft Graph to create/manage Azure resources on behalf of the user. And for that we need to have a application registration which defines authentication flows and permissions it will need. Then it is required that admin of your tenant consents to use of this application within the tenant. This is one-time only action that your tenant admin will have to perform. Only after admin consent is granted deployment application can work in your tenant.

During the application execution, authentication flow will automatically check whether admin consent is granted or not. If not, then after authentication you will be either presented with page to immediately grant the consent (only if you are tenant admin) or to ask your tenant admin to grant consent for use of deployment application (`AzureIndustrialIoTDeployment`) within the tenant. If you are not admin, you should be able to propagate the request to your tenant admin.

Here is documentation of deployment application, it also includes notes about admin consent: https://github.com/Azure/Industrial-IoT/blob/master/docs/industrial_iot_deployment.md

Some additional resources about Azure AD consent:
* [Understanding Azure AD application consent experiences](https://docs.microsoft.com/azure/active-directory/develop/application-consent-experience)
* [Configure the way end-users consent to an application in Azure Active Directory](https://docs.microsoft.com/azure/active-directory/manage-apps/configure-user-consent)

Karapet

karok2m, 2019-11-29:
@mma-mm, I should also mention that currently you would not be able to run `Microsoft.Azure.IIoT.Deployment` if you are not admin in your tenant, even after admin consent is granted to `AzureIndustrialIoTDeployment` application. The reason for that is that `Microsoft.Azure.IIoT.Deployment` does application registration in your tenant, and that is not possible if you are not admin in the tenant. We will add functionality in future to pre-register applications in the tenant and pass application ids to  `Microsoft.Azure.IIoT.Deployment`, so that it uses them instead of creating new application registration.

karok2m, 2020-01-14:
@mma-mm, we've introduced separation of deployment flow via `Microsoft.Azure.IIoT.Deployment` into two stages. First one (`RunMode=ApplicationRegistration`) performs only AAD application setup and second one (`RunMode=ResourceDeployment`) performs deployment of Azure resources and microservices. The first step does still require admin permissions, since they are required for registration of AAD applications, but for the second one being an Owner of a subscription would suffice. Would this separation help you in deploying our IIoT solution ?

Please note that in `RunMode=ApplicationRegistration` application will output definitions of registered applications that should be then used for running in `RunMode=ResourceDeployment`. You can find more info in [Microsoft.Azure.IIoT.Deployment  documentation](https://github.com/Azure/Industrial-IoT/blob/master/docs/industrial_iot_deployment.md). For passing output of first step to second one you can use appsettings.json file. Here is how sample [appsettings.json](https://github.com/Azure/Industrial-IoT/blob/master/deploy/src/Microsoft.Azure.IIoT.Deployment/appsettings.json) should look.

marcschier, 2020-02-19:
Closing for inactivity - current solution requires a manual step by an administrator to consent.   This cannot be avoided at this point.   We are working on a version that is AAD independent which should help in the future.

closed 2020-02-19 by marcschier