Azure/Industrial-IoT #173: Scanning for OPC-Server closes open connection to OPC-Server

mma-mm, 2019-12-13:
**Describe the bug**
Scanning OPC-servers closes an open connection to the OPC-server in the opc-twin-module.

**To Reproduce**
Steps to reproduce the behavior:
1. Deploy the IIoT-Stuff
2. Scan for OPC-servers
3. Connect to an OPC-server (e.g. activate endpoint)
4. Trigger Another Scan for the same OPC-Server (e.g. via twin)
5. See in log how the connection to the OPC-Server gets closed, any browse, read, write, method-call requests to the OPC-Server fail.

Activating again reconnects to the OPC-Server and allows communication. Nevertheless the interruption is disturbing. 

**Expected behavior**
Scanning for OPC-Servers should not interrupt communication.

**Additional Context**
Using OPC-Twin module Version 2.5.1 for AMD64 (Linux)

monikavar, 2020-02-21:
I checked the logs and found that the connection to the OPC-Server does not get closed. It remains connected. Please re-open the issue if the issue persists.

closed 2020-02-21 by monikavar

mma-mm, 2020-02-21:
The error still exists:
    
    [12:30:37 INF] Device uat8bf41a602ea1c30371ed5da9d23a01848693248d connection status changed to Connected due to Connection_Ok.
    [12:30:37 INF] Initialize twin for uat8bf41a602ea1c30371ed5da9d23a01848693248d - standalone
    [12:30:37 INF] 11: Session created.
    [12:30:37 WRN] 11: Establishing unecrypted connection to opc.tcp://172.18.0.2:4840/.
    [12:30:37 INF] 11: Creating session to opc.tcp://172.18.0.2:4840/ via opc.tcp://172.18.0.2:4840/.
    [12:30:37 INF] Module Host started.
    [12:30:37 INF] uat8bf41a602ea1c30371ed5da9d23a01848693248d twin started.
    [12:30:37 INF] New settings processed.
    [12:30:38 INF] 11: Endpoint opc.tcp://172.18.0.2:4840/ via opc.tcp://172.18.0.2:4840/ changed from Connecting to Ready
    [12:32:18 INF] Cancelled discovery.
    [12:32:18 INF] Starting discovery...
    [12:32:18 INF] Stopping Module Host...
    [12:32:18 INF] Start Network discovery run...
    [12:32:18 INF] Start scanning ["172.18.0.0-172.18.0.255 [custom range]"]...
    [12:32:18 INF] Device uat8bf41a602ea1c30371ed5da9d23a01848693248d connection status changed to Disabled due to Client_Close.
    [12:32:18 INF] Module Host stopped.
    [12:32:18 INF] 11: Session closed.
    [12:32:18 INF] uat8bf41a602ea1c30371ed5da9d23a01848693248d twin stopped.
    [12:32:18 INF] New settings processed.
    [12:32:27 INF] Found 9 addresses took 00:00:09.2276723...
    [12:32:27 INF] Probe 0 : [::ffff:172.18.0.2]:4840 found opc server (protocol:0) ...
    [12:32:27 INF] Probe 1 : [::ffff:172.18.0.1]:4840 found opc server (protocol:0) ...
    [12:32:27 INF] Found 2 ports on servers took 00:00:09.3681290...
    [12:32:27 INF] Found 2 endpoints on opcserver.azure-iot-edge:4840 (172.18.0.2).
    [12:32:27 INF] Found 2 endpoints on 172.18.0.1:4840 (172.18.0.1).
    [12:32:27 INF] Discovery took 00:00:09.5286668 and found 1 servers.
    [12:32:27 INF] Uploading 1 results...
    [12:32:27 INF] 1 results uploaded.

In my case Scanning is enabled in the twin with the following properties:
    
    "desired": {
      "AddressRangesToScan": "172.18.0.0/24",
      "Discovery": "Network"
    }

If i do not communicate using the Twin-Service, the scanning occurs every 6 hours. 
As soon as i activate the endpoint and communicate  (read values, call method), this interval seems to be no longer valid. I usually can communicate flawlessly for about 1-2 minutes. Then the scanning kicks in, and i get disconnected.

The problem appears with both amd64 and arm32v7 devices.