Azure/Industrial-IoT #177: OPC Publisher does not report data errors to IoT Hub

RotateAt60MPH, 2019-12-13:
**Is your feature request related to a problem? Please describe.**

**Describe the solution you'd like**
If the read from an OPC Server fails (e.g. the variable was deleted, or the server was unavailable), there should be some indication that something went wrong.

**Describe alternatives you've considered**
None

**Additional context**
Right now, if a specific point is not available, that point is skipped over in the publish to the IoT Hub.    Be more useful if there was some indication that something happened.

barnstee, 2020-09-17:
This is implemented in version 2.7.183.

closed 2020-09-17 by barnstee

RotateAt60MPH, 2020-09-17:
Thanks @barnstee ! Has there been a documentation update on how to leverage this new capability?

barnstee, 2020-09-17:
It is automatic, see [here](https://github.com/Azure/Industrial-IoT/issues/389#issuecomment-627534226)