Azure/Industrial-IoT #184: Publish Future Release Roadmap or Milestones

PWSys, 2020-01-02:
Would it be possible to publish the project roadmap or future releases?  There are a few comments regarding a future release in this quarter.  It would be helpful for the community if we could anticipate releases and their direction.  Ideally, this would specify the next release with anticipated features and timeframe.

marcschier, 2020-01-22:
We are debating to move our feature and bug tracking from our internal DevOps instance to GitHub.   If we do not move - we will provide a roadmap page in our docs.

PWSys, 2020-01-27:
Moving to Github for feature and bug tracking would be great!  Can you share why the team may not decide to do so?

Publishing the road map in the docs would be welcomed as well.

marcschier, 2020-02-12:
Hi, we have decided to not publish a roadmap on github, however, we will share a roadmap with partners.  Also we will keep open feature requests that will be implemented within the next 6 months.

closed 2020-02-12 by marcschier