Azure/Industrial-IoT #199: OPCPublisher module not reporting all Update Events OPCUA server is generating

logaras, 2020-01-17:
Hello, 

I have been using the OPCPublisher edge module to forward events from an OPCUA Server to an IoTHub.
Although the OPCUA Server is generating a fixed number of updates per second (124 nodes with a float variable), IoThub is not receiving that number. 
I verified that my OPCUA server is actually producing that number of events by having a simple OPCUA python client subscribed to the same 
124 nodes.
I had that python client running on identical environment as the OPCPublisher module is running 
(tested both in arm32 and x64 architectures). 
The python client did not miss any update from the server (as opposed to OPCPublisher module)
Server, Clients and Modules running on different VMs or physical machines.

I went back to OPCPublisher logs and verified the following:
- No errors have been reported
- Number of events is always lower than the number or updates the OPCUA Server is generating (and the Python Clients are receiving)

CPU load of the opcpublisher process was max. 24%

Multiple configurations have been tested to tune different parameters (oi,si,ms...) without affecting the issue. 

An example of OPCPublisher diagnostics where the generated events are 7564 but OPCPublisher reports 7322

Configuration: /app/opcpublisher --pf=./pn.json --di=10 --si=10 --oi=0 --ll=debug
[10:35:45 INF] ==========================================================================
[10:35:45 INF] OpcPublisher status @ 01/17/2020 10:35:45 (started @ 01/17/2020 10:34:25)
[10:35:45 INF] ---------------------------------
[10:35:45 INF] OPC sessions (configured/connected): 1/1
[10:35:45 INF] OPC subscriptions (configured/connected): 1/1
[10:35:45 INF] OPC monitored items (configured/monitored/to remove): 124/124/0
[10:35:45 INF] ---------------------------------
[10:35:45 INF] monitored items queue bounded capacity: 8192
[10:35:45 INF] monitored items queue current items: 0
[10:35:45 INF] monitored item notifications enqueued: 7322
[10:35:45 INF] monitored item notifications enqueue failure: 0
[10:35:45 INF] ---------------------------------
[10:35:45 INF] messages sent to IoTHub: 7
[10:35:45 INF] last successful msg sent @: 01/17/2020 10:35:39
[10:35:45 INF] bytes sent to IoTHub: 1551324
[10:35:45 INF] avg msg size: 221617
[10:35:45 INF] msg send failures: 0
[10:35:45 INF] messages too large to sent to IoTHub: 0
[10:35:45 INF] times we missed send interval: 1
[10:35:45 INF] number of events: 7322
[10:35:45 INF] ---------------------------------
[10:35:45 INF] current working set in MB: 108
[10:35:45 INF] --si setting: 10
[10:35:45 INF] --ms setting: 262144
[10:35:45 INF] --ih setting: Amqp_Tcp_Only
[10:35:45 INF] ==========================================================================

**To Reproduce**
Steps to reproduce the behavior:
- Connect to OPCUA Server with OPCPublisher
- Monitor generated OPC UA updates on the server
- Monitor received messages on OPCPublisher
- Number of events received vs. number of events generated differs

**Expected behavior**
OPCPublisher reports as many update events as OPC UA Server is generating

**Environment:**
- Iotedge Runtime 1.0.8 (208b2204fd30e856d00b280112422130c104b9f0) 
- Architecture arm32, x64
- OPCPubilsher version: latest build

ts4iot, 2020-02-10:
From my perspective, the issue describes a critical bug in the OPC-Publisher.
Is there a test case that verfies that all data of an OPC-UA server is fully captured and transmitted to an IoT-Hub?

logaras, 2020-02-26:
Hello, 
I have setup a test infrastructure accessible to you in order to showcase the issue. 
There is an OPCUA server running under opc.tcp://opctestinfra.westeurope.cloudapp.azure.com:4840/opcuatest/server/
that updates 100 variables every second.
The configuration file to connect to the server is here: 

Configuration file for OPCPublisher is here https://www.codepile.net/pile/XwWQYrBl . Just replace EndpointUrl with the url above.

To verify that the server generates the 100 updates per second I use  this simple python client: https://www.codepile.net/pile/OqeAPDq3  
Just replace OPC_URI with the URI above. 
The list with all 100 paths is here https://www.codepile.net/pile/OPAAyBPk . Just save it as paths.txt on the same directory with client.py

The number of received updates on OPC UA client is actually 100 per second:
Python: New data change event 2020-02-26T13:06:14.946829 100
Python: New data change event 2020-02-26T13:06:15.737655 200
Python: New data change event 2020-02-26T13:06:16.235574 300
Python: New data change event 2020-02-26T13:06:17.235307 400
Python: New data change event 2020-02-26T13:06:18.241750 500
Python: New data change event 2020-02-26T13:06:19.399930 600
Python: New data change event 2020-02-26T13:06:20.773507 700
Python: New data change event 2020-02-26T13:06:21.776895 800

While the received events on the OPC Publisher not:
[13:06:18 INF] number of events: 0
[13:06:19 INF] number of events: 100
[13:06:20 INF] number of events: 206
[13:06:21 INF] number of events: 291
[13:06:22 INF] number of events: 387
[13:06:23 INF] number of events: 480
[13:06:24 INF] number of events: 577
[13:06:25 INF] number of events: 669
[13:06:26 INF] number of events: 769
[13:06:27 INF] number of events: 863
[13:06:28 INF] number of events: 956
[13:06:29 INF] number of events: 1048

Please, let me know when you plan to connect to the server yourself so I can make sure the server is up and running.

markusstuhler, 2020-09-23:
Hi @logaras, sorry for the late response from our side.

Is this issue still current?

Thanks!
Markus

logaras, 2020-09-29:
Hi Markus, thanks for replying. 
Please, allow a couple of days to perform some tests and I will reply to you as soon as I have some results. 

Marios

markusstuhler, 2020-10-14:
Hi Marios, 

did you find time to retest if the issue still persists? I could not reproduce in on my side. I wrote a small testing tool that validates the sequence number of the the messages and no message was missing.

Thanks!
Markus

markusstuhler, 2020-10-16:
I will close this issue for now. Please reopen if you're still having problems.

Thanks!
Markus

closed 2020-10-16 by markusstuhler