Azure/Industrial-IoT #216: Integration of IIoT Services - OPC Publisher, Twin and Vault

podgeskeeper, 2020-01-28:
Is there a timeline for implementation/integration of the OPC UA services, OPC Publisher/Twin and Vault?

The documentation references utilization of the services in combination for a complete solution but the integration is not yet available, with specific reference to:

_"The application service is either backed by a CosmosDB 
database or the [OpcTwin device registry (*)](#Yet-Unsupported-features)"_ [in OPC Vault documentation]

What is the plan for the integration of these services?  It would be very useful to be able to manage certificates more autonomously.

marcschier, 2020-01-28:
Hello @podgeskeeper, thank you for your interest.  

We will start by providing support to manage the twin and publisher client certificates centrally through Keyvault.   The work has started and should be part of a preview release some time before June.  

I am marking this item as feature to track the work here on GitHub.

podgeskeeper, 2020-01-29:
Thank you for the update @marcschier.  Is there a plan/timeline to go beyond this and manage certificates for OPC Servers and Clients?

closed 2023-01-16 by marcschier