Azure/Industrial-IoT #217: User 'alias@microsoft.com' returned by service does not match user 'first.last@microsoft.com' in the request

ytechie, 2020-01-28:
**Describe the bug**
When I run `Connect-AzAccount` using my corporate account, it switches the "Account" to my long-form email address. Then, when I run the deploy, the accounts don't match.

**To Reproduce**
Steps to reproduce the behavior:
1. Have a corporate account with both an alias version and a long-form version. In my case this looks like `alias@microsoft.com` and `first.last@microsoft.com`.
1. Run `Connect-AzAccount` and log in with your corporate account. Even if you use `alias@microsoft.com`, the cmdlet will show `first.last@microsoft.com` as the output.
1. Run `./deploy.cmd`
1. When the auth prompt shows up, it will not let you use `first.last@microsoft.com`, so you have no choice but to use `alias@microsoft.com`
1. See the error "User 'alias@microsoft.com' returned by service does not match user 'first.last@microsoft.com' in the request"

**Expected behavior**
There should be a way to get the ID's to match, or even better, it should just work by default.

**Screenshots**
[image]

[image]

[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - PowerShell

ytechie, 2020-01-29:
I tried the new deployment executable in [src/Microsoft.Azure.IIoT.Deployment](https://github.com/Azure/Industrial-IoT/tree/master/deploy/src/Microsoft.Azure.IIoT.Deployment). It didn't have the same authentication error.

karok2m, 2020-03-19:
Hi @ytechie. This should be addressed with https://github.com/Azure/Industrial-IoT/pull/332. Can you please check if latest `master` works for you ?

Please note that you might be prompted to log in twice. This is because we first call `Connect-AzAccount` and then use returned account details for `Connect-AzureAD`. If `Connect-AzureAD` with received `Account.Id` fails (which in your case is `jason.young@microsoft.com`) then we try without it. Hopefully this should solve the problem.

closed 2020-03-27 by marcschier