Azure/Industrial-IoT #221: Deadband in OPC Publisher

FMauroy, 2020-01-30:
Hello, are there plans to allow setting deadband values for each monitored OPC node in OPC Publisher? Our customer would like to avoid sending values that do not change more than a fixed/relative range of values...
I see in the UA API that deadband settings exist, but they are not used in OPC Publisher.
Thanks a lot!

marcschier, 2020-01-31:
Hi @FMauroy  - preview (i.e. master) supports deadband and other filter settings, however, there is no API yet to control this - either through REST or the legacy publishednodes.json file.   This is on our backlog.   Keeping this as feature item to track

FMauroy, 2020-01-31:
Thanks for your quick reply. We'll wait then, no problem.
Keep up the good work :-)

closed 2020-01-31 by FMauroy

pratson20, 2021-09-13:
Hi, i am also looking for dead-band functionality as well. is that support available now?

Also OPC publisher to support synchronous communication?

Regards,
Prateek