Azure/Industrial-IoT #228: Attempt to Register Server -> Returns Nothing when Getting Applications

msha1026, 2020-02-05:
I have deployed the AKS version of the Industrial IoT stack and also an edge VM that has discovery, publisher and twin modules ([following these directions](https://github.com/Azure/Industrial-IoT/blob/master/docs/deploy/howto-deploy-modules-portal.md)). I have followed [these steps](https://github.com/Azure/Industrial-IoT/blob/master/docs/howto-use-postman.md) to use Postman to work with the microservices and simulate a PLC server on my edge VM. I get 200 OK when I register my server; however, I never see the application when doing GET applications. Any guidance on why this may be happening?

I logged some of the telemetry messages coming from IoT Hub to see what was happening and would receive a message with the following: ```..."Event":"EndpointsDiscoveryStarted","Workers":null,"Progress":null,"Discovered":1,"Result":null}82020-02-05T02:42:34.9830000Z```

and another message like this:
```..."Event":"EndpointsDiscoveryFinished","Workers":null,"Progress":null,"Discovered":0,"Result":null}82020-02-05T02:42:34.9980000Z```

Would appreciate any explanation into what these messages mean since Discovered goes from 1 to 0. Thanks!

marcschier, 2020-02-06:
Hi @msha1026, we will look into it ASAP.  Can I ask, which version of publisher and twin are you deploying into IoT Edge?  Also, would you mind providing me with logs from the twin, if you want offline?   Thanks so much!

msha1026, 2020-02-06:
Hi @marcschier , thanks a lot! I'm pulling the latest modules from MCR for publisher and twin. I can get some logs for the twin to you by EOD. Thanks again!

msha1026, 2020-02-06:
@marcschier I have sent you an email with the logs from my twin module. Also noted in the email that we were able to get this to work when rolling back to 2.5.2 edge modules. I appreciate any guidance you have around this. Thanks so much!

marcschier, 2020-02-18:
Edge modules need to be version aligned with the cloud services.   Edge manager takes care of it by updating the layered deployment for all IIoT Modules, so the module container tag is the same as the version of the edge manager.   This means 2.6 cloud services align with 2.6 modules, but not 2.5 (for all scenarios - i.e. discovery).   Closing.

closed 2020-02-18 by marcschier