Azure/Industrial-IoT #245: Autorest Generated Client Incorrect

msha1026, 2020-02-12:
**Describe the bug**
I have generated a Java client library using autorest and the swagger specs in this repo. However, the autorest generated code seems to be stuffing the request headers with info that the microservices cannot handle.

For example the following snippet of code generated by autorest is what calls the IIoT microservices:
```
@Headers({ "Content-Type: application/json; charset=utf-8", "x-ms-logging-context: com.microsoft.azure.iiot.ua.sdk.swagger.registry.OpcRegistryService activateEndpoint" })
@POST("v2/endpoints/{endpointId}/activate")
Observable<Response<ResponseBody>> activateEndpoint(@Path("endpointId") String endpointId); 
```

Initially, calling activateEndpoint failed and gave me errors like: `The format of value 'application/json; charset=utf-8' is invalid.` 
After removing `"Content-Type: application/json; charset=utf-8"` from the code snippet above, activateEndpoint worked as expected.

**To Reproduce**
Steps to reproduce the behavior:
1. Use autorest to generate Java client library
2. Call activateEndpoint
3. See error: 'application/json; charset=utf-8' is invalid.` 

**Expected behavior**
Autorest should generate correct client library off of swagger specs or microservices should handle what the generated client code is sending to them.

**Additional context**
Also confirmed through Postman, that calling activate endpoint with `Content-Type: application/json; charset=utf-8` in the Headers resulted in the same problem.

msha1026, 2020-02-12:
I also wanted to note that this error occurs with deactivateEndpoint as well. These are the only two calls I have noticed that have these issues thus far. Unsure if there are more producing this problem since I have only tested a very small subset (getListOfEndpoints, getListOfApplications, de/activateEndpoint).

marcschier, 2020-02-18:
Hi @msha1026 - this likely occurs for all API.   We are going to fix this in sprint 2003.

msha1026, 2020-02-21:
@marcschier thanks for the info :) What does sprint 2003 mean in regards to date? Is there an approximate date?

monikavar, 2020-03-04:
Hi @msha1026 , I tried to reproduce for the same API call with Postman, however, it was working fine. Screenshot below:
[image]
Could you please try again with the newer codebase and also attach your swagger json file which you used to generate Java client using autorest?

msha1026, 2020-03-06:
@monikavar I can confirm I do not see this error with the new release. The swagger json I was using was with the 2.5.2 release.

closed 2020-03-06 by msha1026