Azure/Industrial-IoT #258: REST API - No endpoint available using deployment-aks

JanuszTaterka, 2020-02-17:
Hi,
I tried to deployed the solution on AKS.
**All components have been successfully deployed**. The swagger documentation is available e.g. under *name*.azurewebsites.net/twin/. However, every service call e.g. https://appname.azurewebsites.net/v2/status, ends up with an 404 error.

Could you please check if the deployment script works as expected?

karok2m, 2020-02-20:
Hi @JanuszTaterka,

You are correct, this is unfortunately a known bug that we are working on fixing. The problem is that swagger of each microservice is not aware of the reverse proxy that it is running behind. So it does not know that endpoint for getting status of registry service is not on `/v2/status`, but rather `/registry/v2/status`.

There are a few workarounds that I can suggest at the moment:

1. If you are calling API endpoints manually, then please note that each microservice is on a specific endpoint, i.e. registry service is under `/registry`, twin is under `/twin` and so on. So to access status of registry service, the endpoint that you need to call is  `https://appname.azurewebsites.net/registry/v2/status`.

2. Use CLI tool. Please note that `Microsoft.Azure.IIoT.Deployment` deploys `2.5.2` version of our components, so you would have to use [`release/2.5.2`](https://github.com/Azure/Industrial-IoT/tree/release/2.5.2) branch. The CLI tool is in `api\src\Microsoft.Azure.IIoT.OpcUa.Api\cli\` directory. You would have to build and run it yourself. To point it to your deployment, please put `.env` file generated by the deployment application in the root directory of the repository.

3. In case of AKS deployments, you can utilize kubectl [port forwarding](https://kubernetes.io/docs/tasks/access-application-cluster/port-forward-access-application-cluster/) to proxy the service endpoint to your local machine. Setting up port forwarding is quite straightforward but to make authentication work in that setup you would need to do a few changes to App Registration of clients application to let you authenticate with `localhost` endpoints. Please let me know if you need more details on this.

karok2m, 2020-03-05:
Hi @JanuszTaterka. This issue has been fixed by https://github.com/Azure/Industrial-IoT/pull/269 for our cloud microservices. Unfortunately, deployment to AKS is still using **2.5.2** version of cloud microservices, so the fix is not visible there yet. We will upgrade AKS deployment to 2.6.x version soon. In the mean time you can try [all-in-one deployment](https://github.com/Azure/Industrial-IoT/blob/master/docs/deploy/howto-deploy-all-in-one.md), where Swagger should work properly now.

closed 2020-03-09 by marcschier