Azure/Industrial-IoT #291: OPC publisher module rPI 4 - "exec format error"

RotateAt60MPH, 2020-03-02:
I am trying to get the OPC Publisher working on a rPI4 with Raspian Buster.  After the module is deployed, it does not start and has the following error:

> pi@raspberrypi:~ $ sudo iotedge logs publisher -f
> `standard_init_linux.go:211: exec user process caused "exec format error"`

And once iotedge is running, I have verified that the main iotedge modules are running:

> pi@raspberrypi:~ $ sudo iotedge list
> NAME             STATUS           DESCRIPTION      CONFIG
> edgeHub          running          Up 20 minutes    mcr.microsoft.com/azureiotedge-hub:1.0
> edgeAgent        running          Up 29 minutes    mcr.microsoft.com/azureiotedge-agent:1.0

'sudo iotedge check' runs with no errors.

Docker was installed using the get-docker script.  

> pi@raspberrypi:~ $ curl -fsSL https://get.docker.com -o get-docker.sh
> pi@raspberrypi:~ $ sudo sh get-docker.sh

I've verified that docker is running since I am able to see the containers:

> pi@raspberrypi:~ $ docker container list
> CONTAINER ID        IMAGE                                      COMMAND                   CREATED             STATUS              PORTS                                                                  NAMES
> 2e49ec44dfb3        mcr.microsoft.com/azureiotedge-hub:1.0     "/bin/sh -c 'echo \"$â¦"   20 minutes ago      Up 20 minutes       0.0.0.0:443->443/tcp, 0.0.0.0:5671->5671/tcp, 0.0.0.0:8883->8883/tcp   edgeHub
> 3831e38148ec        mcr.microsoft.com/azureiotedge-agent:1.0   "/bin/sh -c 'echo \"$â¦"   29 minutes ago      Up 29 minutes   

If it was the wrong build of docker (X64 versus armhf), I wouldn't see the edge containers would I?

Any ideas appreciated.

RotateAt60MPH, 2020-03-03:
Further information.  I stopped the iotedge service, and deleted the publisher container in docker.  I then went back to OPC Publisher 2.5.3 version and the problem went away.

Certainly looks like somebody got broke something in the :latest version.

cristipogacean, 2020-03-03:
Thanks for reporting this, we will look into it asap!

cristipogacean, 2020-03-04:
@RotateAt60MPH, I just tested 2.6 and it worked fine on my raspberry with the :latest publisher, twin and discoverer containers.:

pi@iiot-arm-host5:~ $ sudo iotedge list
NAME             STATUS           DESCRIPTION      CONFIG
edgeHub          running          Up a day         mcr.microsoft.com/azureiotedge-hub:1.0
discovery        running          Up 21 hours      mcr.microsoft.com/iotedge/discovery:2.6.96
edgeAgent        running          Up a day         mcr.microsoft.com/azureiotedge-agent:1.0
publisher        running          Up 21 hours      mcr.microsoft.com/iotedge/opc-publisher:2.6.96
twin             running          Up 21 hours      mcr.microsoft.com/iotedge/opc-twin:2.6.96

I did setup the iotedge environment on the raspberry PI following: https://docs.microsoft.com/en-us/azure/iot-edge/how-to-install-iot-edge-linux

the differences between our test scenarious are:
1) I use a Raspebrry PI 3 - but this should normally make no difference
2) I use moby as described in the previously mentioned how-to-doc not docker from docker.com
3) I used a fresh azure deployment based on 2.6 scripts ad I added a new edge device in the fresh iothub

Would it work for you to try tuning your environment in the direction I tested?

RotateAt60MPH, 2020-03-04:
@cristipogacean I had used that page as my original guide for installing IoT on the PI 4.  However, at the time, about 6 weeks ago, moby would not install on the PI 4.  There were errors.  This would have been with the pre-2.6 iotedge, I'm assuming.

I will run another test - perhaps iot 2.6 works better with moby.  I will report the results.

FYI, I've found that there are a lot of scenarios where configurations or software that works on the PI 3 will not work on the PI 4.  It's that much different.

RotateAt60MPH, 2020-03-05:
Hello @cristipogacean. 

I tried the install using the exact directions in the webpage you suggested (using moby, instead of docker-ce as I did in my original test) and the OPC Publisher once more failed to start.  At least it is consistent.

Same error:

`
pi@raspberrypi:~ $ sudo iotedge logs publisher -f
standard_init_linux.go:211: exec user process caused "exec format error"
standard_init_linux.go:211: exec user process caused "exec format error"
standard_init_linux.go:211: exec user process caused "exec format error"
standard_init_linux.go:211: exec user process caused "exec format error"
pi@raspberrypi:~ $
`

I will put aside this SD card for awhile, if you need me to provide more information.

cristipogacean, 2020-03-05:
thanks a lot for the feedback! then we need to get a Rpy 4,replicate your scenario and look into the cause of the docker regression from 2.5.3

RotateAt60MPH, 2020-03-05:
@cristipogacean One final tidbit of info - deploying OPC Publisher 2.5.3 to this same rPI 4 worked - allowed the module to start up.  Similar to my test previously.

marcschier, 2020-03-06:
This looks like an issue in our multi arch images and which layer is selected on pi3 vs pi4.  Seems on pi4 amd64 is selected.

marcschier, 2020-03-19:
Hi @RotateAt60MPH - we added ARM64 (linux/arm64) layer to our images.   In 2.3x we used linux/arm, which might have resolved to work on pi4, on 2.5x we used linux/arm/v7, which probably did not work correctly anymore.  In 2.6 we added several Windows layers including Windows/arm, and that could have caused it.   Since we do not have an Rpi4 handy, could you try the latest "preview" tagged module and see if it works for you now?   Please reopen if not.

closed 2020-03-19 by marcschier

JohnnyCrazy, 2020-04-02:
Hi,

I can confirm the preview images work on a `Moxa UC-8100A`, `ARMv7`.
`Linux Moxa 4.4.0-cip-uc8100a-me #12 Tue Dec 11 14:53:48 CST 2018 armv7l GNU/Linux`

When can we expect the `preview` images to replace `latest` @marcschier ?

DanielBott, 2021-03-31:
I am trying to get the OPC Publisher working on a PI4 with Ubuntu 18.04. After the module is deployed, it does not start and has the following error:
```
ubun1804@Rasp4:~$ sudo iotedge logs OPCPublisher
standard_init_linux.go:219: exec user process caused: no such file or directory
```

the main iotedge modules are running:
```
ubun1804@Rasp4:~$ sudo iotedge list
NAME             STATUS           DESCRIPTION               CONFIG
OPCPublisher     failed           Failed (1) 2 minutes ago  mcr.microsoft.com/iotedge/opc-publisher:latest
edgeAgent        running          Up 4 hours                mcr.microsoft.com/azureiotedge-agent:1.0
edgeHub          running          Up 4 hours                mcr.microsoft.com/azureiotedge-hub:1.0
```

in my point of view the iotedge is running very well
```
ubun1804@Rasp4:~$ sudo iotedge check
Configuration checks
--------------------
√ config.yaml is well-formed - OK
√ config.yaml has well-formed connection string - OK
√ container engine is installed and functional - OK
√ config.yaml has correct URIs for daemon mgmt endpoint - OK
√ host time is close to real time - OK
√ container time is close to host time - OK
√ production readiness: container engine - OK
Connectivity checks
-------------------
√ host can connect to and perform TLS handshake with IoT Hub AMQP port - OK
√ host can connect to and perform TLS handshake with IoT Hub HTTPS / WebSockets port - OK
√ host can connect to and perform TLS handshake with IoT Hub MQTT port - OK
√ container on the default network can connect to IoT Hub AMQP port - OK
√ container on the default network can connect to IoT Hub HTTPS / WebSockets port - OK
√ container on the default network can connect to IoT Hub MQTT port - OK
√ container on the IoT Edge module network can connect to IoT Hub AMQP port - OK
√ container on the IoT Edge module network can connect to IoT Hub HTTPS / WebSockets port - OK
√ container on the IoT Edge module network can connect to IoT Hub MQTT port - OK
16 check(s) succeeded.
```
the setup followed https://docs.microsoft.com/en-us/azure/iot-edge/how-to-install-iot-edge?view=iotedge-2018-06 

my intallation runs since some days
```
ubun1804@Rasp4:~$ sudo docker container list
CONTAINER ID   IMAGE                                      COMMAND                   CREATED       STATUS       PORTS                                                                  NAMES
10611d7fb73f   mcr.microsoft.com/azureiotedge-hub:1.0     "/bin/sh -c 'echo \"$…"   12 days ago   Up 5 hours   0.0.0.0:443->443/tcp, 0.0.0.0:5671->5671/tcp, 0.0.0.0:8883->8883/tcp   edgeHub
8143f0b6ba86   mcr.microsoft.com/azureiotedge-agent:1.0   "/bin/sh -c 'exec /a…"    12 days ago   Up 5 hours                                                                          edgeAgent
```

also 
(1) I tried to deploy other edge device modules and it works. e.g. a demo module written by myself or the azure-iot.simulated-temperature-sensor module
(2) I also deployed other OPC Publisher Verions 2.5.3 and 2.3x. e.g. 2.5.3:
```
ubun1804@Rasp4:~$ sudo iotedge logs edgeAgent
<6> 2021-03-31 10:58:53.157 +00:00 [INF] - Plan execution ended for deployment 25
<3> 2021-03-31 10:58:53.157 +00:00 [ERR] - Edge agent plan execution failed.
System.AggregateException: One or more errors occurred. (Error calling prepare update for module OPCPublisher: Could not prepare update for module "OPCPublisher"
        caused by: Could not pull image mcr.microsoft.com/iotedge/opc-publisher:2.5.3
        caused by: no matching manifest for linux/arm64/v8 in the manifest list entries)
 ---> Microsoft.Azure.Devices.Edge.Agent.Edgelet.EdgeletCommunicationException- Message:Error calling prepare update for module OPCPublisher: Could not prepare update for module "OPCPublisher"
        caused by: Could not pull image mcr.microsoft.com/iotedge/opc-publisher:2.5.3
        caused by: no matching manifest for linux/arm64/v8 in the manifest list entries, StatusCode:500, at:   at Microsoft.Azure.Devices.Edge.Agent.Edgelet.Version_2020_07_07.ModuleManagementHttpClient.HandleException(Exception exception, String operation) in /home/vsts/work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Edgelet/version_2020_07_07/ModuleManagementHttpClient.cs:line 232
   at Microsoft.Azure.Devices.Edge.Agent.Edgelet.Versioning.ModuleManagementHttpClientVersioned.Execute[T](Func`1 func, String operation) in /home/vsts/work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Edgelet/versioning/ModuleManagementHttpClientVersioned.cs:line 143
   at Microsoft.Azure.Devices.Edge.Agent.Edgelet.Version_2020_07_07.ModuleManagementHttpClient.PrepareUpdateAsync(ModuleSpec moduleSpec) in /home/vsts/work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Edgelet/version_2020_07_07/ModuleManagementHttpClient.cs:line 214
   at Microsoft.Azure.Devices.Edge.Agent.Core.Commands.GroupCommand.ExecuteAsync(CancellationToken token) in /home/vsts/work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Core/commands/GroupCommand.cs:line 35
   at Microsoft.Azure.Devices.Edge.Agent.Core.LoggingCommandFactory.LoggingCommand.ExecuteAsync(CancellationToken token) in /home/vsts/work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Core/LoggingCommandFactory.cs:line 67
   at Microsoft.Azure.Devices.Edge.Agent.Core.Commands.GroupCommand.ExecuteAsync(CancellationToken token) in /home/vsts/work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Core/commands/GroupCommand.cs:line 35
[... cut ...]
   at Microsoft.Azure.Devices.Edge.Agent.Core.Agent.ReconcileAsync(CancellationToken token) in /home/vsts/work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Core/Agent.cs:line 203
```
(3) it makes no difference if I use: 
```
mcr.microsoft.com/iotedge/opc-publisher:latest or 
mcr.microsoft.com/iotedge/opc-publisher:preview
```
 i get following error:
```
ubun1804@Rasp4:~$ sudo iotedge logs OPCPublisher
standard_init_linux.go:219: exec user process caused: no such file or directory
```

Do you have any idea what it can be?