Azure/Industrial-IoT #292: Unable to GET applications using API for 2.6 IIoT

jhastings88, 2020-03-04:
**Description**
I am using Postman to GET applications at the edge. When I make the request to discover applications, the request is successful. I can also look at the discovery module’s logs at on the edge to verify it’s returning a discovered application. 

The Discover module's logs from the edge device
```
[20:34:03 INF] Discovery took 00:00:18.3995031 and found 1 servers. (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.DiscoveryServices)
[20:34:03 INF] Uploading 1 results... (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.DiscoveryServices)
[20:34:03 INF] 1 results uploaded. (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.DiscoveryServices)

```

But when I try to GET applications, the list is empty. I'm not sure where that data isn’t getting though. We also looked at the app service log stream and we’re seeing an exception getting thrown and the details are something is not getting authenticated correctly.

**How I deployed the 2.6 IIoT stack**
In case this offer's any clues I'll describe how I deployed 2.6.
First I created my own resource group, then I ran **aad-register.ps1 -Name “myResourceGroupName”** in the script directory. I do this because I need to run the deploy script in the context of a service principal and there is a azure cli command that gets called **Connect-AzAccount** in the aad-register.ps1 script that I can't figure out how to run in the context of a service principal. For some reason I can’t login in as a service principal using that command. So instead I run the aad-register script as myself to create my app registrations. Then in the context of my service principal I call **deploy.ps1 -aadConfig “pathToMyConfigFileReturnedFromAadRegisterScript”.** 

After the deployment is successful, in the powershell I see
```
The deployed application can be found at:
https://myResourceGroupName.azurewebsites.net
To be able to use the application you need to register the following
reply url for AAD application ###################:
https://myResourceGroupName.azurewebsites.net/signin-oidc
```
It also asks if I want to save my environment, but that’s all it says after it’s deployed. 

In my app registration for the client I see there are no reply URLs so I add the one it suggested. The **https://myResourceGroupName.azurewebsites.net/signin-oidc** reply URL. I’m able to reach https://myResourceGroupName.azurewebsites.net website just fine.
As part of the tut-use-postman.md tutorial I also add https://myResourceGroupName.azurewebsites.net/.auth/login/aad/callback to the reply URL for the client app registration.

**Expected behavior**
I expected the GET applications request to return an application

marcschier, 2020-03-04:
Hi @jhastings88 , could you add the exception and context from the log?

jleger-ptc, 2020-03-05:
Hi @marcschier , I am also seeing a similar issue to @jhastings88 in a deployment that was performed via the deploy.ps1 script with **type = services**. 

Following the successful deployment, an Edge VM was provisioned using the [Azure IoT Edge on Ubuntu](https://azuremarketplace.microsoft.com/en-us/marketplace/apps/microsoft_iot_edge.iot_edge_vm_ubuntu?tab=overview) image and by following the [howto-deploy-modules-portal](https://github.com/Azure/Industrial-IoT/blob/master/docs/deploy/howto-deploy-modules-portal.md) tutorial.

After confirming that the supervisor was connected and discoverable via Postman, a "Local" application discovery was initiated and resulted in the following log output from the discovery module at the edge

> [04:43:27 INF] null: Discovery operation started. (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher)
> [04:43:27 INF] Start Local discovery run... (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.DiscoveryServices)
> [04:43:27 INF] null: Starting network scan (0 probes active)... (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher)
> [04:43:27 INF] Start scanning ["172.17.4.4/32 [eth0]", "172.19.0.1/32 [br-8c729debdf4c]", "172.18.0.1/32 [docker0]"]... (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.DiscoveryServices)
> [04:43:27 INF] null: Found address 172.17.4.4 (0 scanned)... (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher)
> [04:43:27 INF] null: Found address 172.19.0.1 (1 scanned)... (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher)
> [04:43:27 INF] null: Found address 172.18.0.1 (2 scanned)... (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher)
> [04:43:27 INF] null: Found 3 addresses. (3 scanned)... (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher)
> [04:43:27 INF] null: Starting port scanning (0 probes active)... (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher)
> [04:43:30 INF] Probe 623 : found OPC UA server at [::ffff:172.17.4.4]:49330 (protocol:0) ... (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.DiscoveryServices)
> [04:43:30 INF] null: Found server 172.17.4.4:49330 (51078 scanned)... (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher)
> [04:43:32 INF] GC Mem: 18740 kb, Working set / Private Mem: 141904 kb / 514600 kb, Handles: 1183 (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.DiscoveryServices)
> [04:43:32 INF] null: 93138 ports scanned - 1 discovered (1000 probes active)... (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher)
> [04:43:34 INF] Probe 984 : found OPC UA server at [::ffff:172.19.0.1]:49330 (protocol:0) ... (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.DiscoveryServices)
> [04:43:34 INF] null: Found server 172.19.0.1:49330 (119373 scanned)... (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher)
> [04:43:35 INF] null: 146213 ports scanned - 2 discovered (1000 probes active)... (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher)
> [04:43:37 INF] Probe 196 : found OPC UA server at [::ffff:172.18.0.1]:49330 (protocol:0) ... (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.DiscoveryServices)
> [04:43:37 INF] null: Found server 172.18.0.1:49330 (183787 scanned)... (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher)
> [04:43:38 INF] null: 196608 ports scanned - 3 discovered (1 probes active)... (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher)
> [04:43:41 INF] null: Found 3 ports on servers (196608 scanned)... (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher)
> [04:43:41 INF] null: Searching 3 discovery urls for endpoints... (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher)
> [04:43:41 INF] null: Trying to find endpoints on opc.tcp://tke-edge-device.internal.cloudapp.net:49330/... (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher)
> [04:43:42 INF] null: Found 1 endpoints on opc.tcp://tke-edge-device.internal.cloudapp.net:49330/. (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher)
> [04:43:42 INF] null: Trying to find endpoints on opc.tcp://172.19.0.1:49330/... (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher)
> [04:43:43 INF] null: Found 1 endpoints on opc.tcp://172.19.0.1:49330/. (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher)
> [04:43:43 INF] null: Trying to find endpoints on opc.tcp://172.18.0.1:49330/... (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher)
> [04:43:44 INF] null: Found 1 endpoints on opc.tcp://172.18.0.1:49330/. (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher)
> [04:43:44 INF] null: Found total of 1 servers ... (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher)
> [04:43:44 INF] Discovery took 00:00:17.3087725 and found 1 servers. (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.DiscoveryServices)
> [04:43:44 INF] Uploading 1 results... (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.DiscoveryServices)
> [04:43:44 INF] 1 results uploaded. (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.DiscoveryServices)
> [04:43:44 INF] null: Discovery operation completed. (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher)

[... cut ...]
> <4> 2020-03-05 04:43:32.039 +00:00 [WRN] - TLS handshake failed., DotNetty.Handlers.Tls.NotSslRecordException: not an SSL/TLS record: 48454c463900000000000000f0ff0f00f0ff0f00f0ff0f00f0ff0f00190000006f70632e7463703a2f2f3137322e31392e302e313a38383833, 606c9c8b
> <6> 2020-03-05 04:43:34.968 +00:00 [INF] - HttpExtensionConnectionAdapter authentication failed
> <4> 2020-03-05 04:43:35.422 +00:00 [WRN] - TLS handshake failed., DotNetty.Handlers.Tls.NotSslRecordException: not an SSL/TLS record: 48454c463900000000000000f0ff0f00f0ff0f00f0ff0f00f0ff0f00190000006f70632e7463703a2f2f3137322e31382e302e313a38383833, 021ded39

jleger-ptc, 2020-03-05:
After deploying a linux gateway via the deploy.ps1 script, I attempted to diff the IoT Hub and IoT Edge configurations between the two deployments.

One thing I noticed is that in the **Device Twin** for the IoT Edge device on the hub, there was the following  "tags":

`"tags": {
    "os": "Linux",
    "__type__": "iiotedge"
  }`

These tags appear to be called out as part of the [howto-install-iot-edge](https://github.com/Azure/Industrial-IoT/blob/master/docs/deploy/howto-install-iot-edge.md#deploy-the-industrial-iot-workloads-to-the-gateway) which was skipped due to starting with the **Azure IoT Edge on Ubuntu** image.

The other difference I saw was the gateway was deploying specific version modules **(2.6.100)** instead of **latest** as called out in the [tutorial](https://github.com/Azure/Industrial-IoT/blob/master/docs/deploy/howto-deploy-modules-portal.md). I wasn't able to find the container registry, so I was unable to confirm if these resolve to the same version or not.

After manually making these changes on my device, I was able to perform **Local** application discovery and confirm that it was returned in the Postman **Get Applications** request.

Perhaps important to note though, I am still seeing the EdgeHub module reporting the authentication/TLS issues upon every discovery event

> <6> 2020-03-05 15:53:08.678 +00:00 [INF] - HttpExtensionConnectionAdapter authentication failed
> <4> 2020-03-05 15:53:09.246 +00:00 [WRN] - TLS handshake failed., DotNetty.Handlers.Tls.NotSslRecordException: not an SSL/TLS record: 48454c463900000000000000f0ff0f00f0ff0f00f0ff0f00f0ff0f00190000006f70632e7463703a2f2f3137322e31392e302e313a38383833, 4da90147

monikavar, 2020-03-18:
Hi @jhastings88 I tried to replicate the steps you mentioned in the description to deploy IIot Stack and tried with Postman to discover and get applications. However, in my case, I don't see an issues. I could see the responses of HTTP POST/GET calls.
[image]
Could you please try it again with the latest codebase to make sure it works for you?

closed 2020-03-29 by marcschier