Azure/Industrial-IoT #296: Using Device Twin Tags Deploys Incorrectly Named Edge Modules

msha1026, 2020-03-05:
**Describe the bug**
When deploying 2.6 IIoT Stack, we add the following tags to our device twin to ensure the correct edge modules are deployed to our gateway device:
```
"tags": {
    "__type__": "iiotedge",
    "os": "Linux"    
}
```
However, when trying to hit the rest endpoints like activate an endpoint, we have this error:
```
ErrorCode:ModuleNotFound;Device <device-name> Module opctwin at IotHub <iothub-name> not registered.
```
Because of this error, I noticed the automatically deployed edge modules were named `twin` and `publisher` but my error message seemed to signal that these edge modules need to be named `opctwin` and `opcpublisher`. Once changing the names, activating endpoint works but then after a few minutes, the edge modules get named back to `twin` and `publisher`. I am assuming this is because the tags are checking that the edge modules are named `twin` and `publisher`.

**Expected behavior**
Using tags in device twin will deploy correctly named modules to the gateway.

cristipogacean, 2020-03-06:
Can you give us some details on how are you trying to activate the endpoint? are using the cli, or swagger or the Engineering tool?

msha1026, 2020-03-06:
We are using Postman. Following the Postman tutorial you have in your docs.

marcschier, 2020-03-06:
This looks like an issue in our docs regarding name of the module identity.

marcschier, 2020-03-16:
Closing as no repro.

closed 2020-03-16 by marcschier