Azure/Industrial-IoT #297: Endpoint Activation: Never goes to ActivatedAndConnect and Ready State

msha1026, 2020-03-06:
**Describe the bug**
When activating an endpoint through Postman, the `activationState` never changes to `ActivatedAndConnected` and `endpointState` does not change to `Ready`.
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Deploy 2.6 IIoT Stack Version
2. Follow tutorial to get setup on postman
3. Activate an endpoint
4. activationState never changes out of Activated

**Expected behavior**
Expect to see `activationState` and `endpointState` to look like the following after a minute or less:
[image]

**Screenshots**
Here are logs pulled from the opctwin module too when trying to activate an endpoint:
[image]

cristipogacean, 2020-03-06:
Do you use a secure endpoint when connecting to the server? if so, probably the server needs to be configured to trust the twin's certificate  requires the twin's certificate. Can you check if the server has an endpoint with security policy none and try that out?

msha1026, 2020-03-06:
I have already trusted the twin’s certificate on the server.

marcschier, 2020-03-06:
It looks like connection to iot hub (appears you run outside of iot edge) is failing Everytime the module is trying to report the connection ready state.  Possible that there are 2 twin modules connecting to via the same identity?

msha1026, 2020-03-06:
Hm interesting @marcschier . Is there a way to check this? When I do an `iotedge list`, I only see one instance of the twin module.

msha1026, 2020-03-06:
When restarting the twin module, activation/deactivation works once, but then does not work again without having to restart the twin module again.

marcschier, 2020-03-07:
It would be super helpful to have a step by step repro, e.g. how was the gateway installed?  Os version, edge version, etc.  As the simulated edges do not exhibit this issue it would help to isolate the difference to our simulated edge deployments.

msha1026, 2020-03-08:
Hi Marc, FYI we are using the Linux Gateway that gets deployed with the all in one deployment and using our own PLC container. All edge modules are pulling from the 2.6.1 tag. I can get a full step by step repro to you on Monday.

msha1026, 2020-03-10:
Hi @marcschier , I will be emailing you a full step-by-step repro soon. Please be on the lookout. Thanks!

marcschier, 2020-03-11:
Hi @msha1026 - got it - we will repro.   One bug is that the documentation for postman tutorial is provides a wrong path for the publisher API.  We shall change that to /publisher.  But the activation state and endpoint state is something that we have so far not seen.  To isolate while we reproduce I would like you to also try out against the full simulation instead of creating your own IoT Edge VM, would that be possible?

marcschier, 2020-03-19:
Hi @msha1026 - I just checked in a activation scenario test - tag 2.6.119.  I was able to reproduce race conditions in the twin module and iot hub sdk on "close", between UpdateProperties and CloseAsync which can lead to unresponsiveness of the module.  This is probably triggered somehow by high load on the module for other work, such as browsing.   Without it I can easily get into this state after around 30 activate/browse/deactivate cycles.  With the fixes I ran 1000 cycles on all connected sessions without seeing issues.   Very early, but could you give 2.6.119 a try while I do more testing?

msha1026, 2020-03-19:
Hi @marcschier, after some tests with 2.6.119 modules, we are still seeing the same issues of the endpoint hanging in `Activated` after a deactivate/activate cycle with TKE. We also notice browsing will push it into this hung `Activated` state as well.

msha1026, 2020-03-24:
@marcschier I have spun up another deployment on Friday which uses the 2.6.122 modules. We have not seen the hung Activated state yet which is promising. We will continue testing and let you know if the issue arises again.

marcschier, 2020-03-25:
@msha1026, greatly appreciated.  There is another fix just pushed today where the endpoint state is not updated from disconnected even though the state is != disconnected on the twin.   This happens when you activate, deactivate, activate.

marcschier, 2020-03-29:
We believe this is fixed.  Please reopen if you still see issues.

closed 2020-03-29 by marcschier