Azure/Industrial-IoT #302: Browse results incompatible with Kepware UA Server

pcameron-ptc, 2020-03-10:
**Describe the bug**
Browse through swagger generated client returns ExpandedNodeId formatted node IDs. This is problematic for brownfield Kepware UA servers. 

**To Reproduce**
Steps to reproduce the behavior:
1. Deploy the Industrial IoT stack
2. Use autorest to generate Java swagger client using a command like the following:
`autorest --input-file=swagger/publisher.json --output-folder=./' --namespace=<xyz> --java
3. Create a rest client, passing the appropriate secrets and credentials
4. Invoke browse using `OpcTwinService.browse` on a Channel node in a "ThingWorx Kepware Edge" installation (i.e. "ns=2;s=Channel1")
5. The browse result (a "Device" node) node ID is returned in the "ExpandedNodeId" format: 

 {
      "referenceTypeId": "i=35",
      "direction": "Forward",
      "target": {
        "nodeClass": "Object",
        "displayName": "Device1",
        "nodeId": "nsu=ThingWorxKepwareEdge;s=Channel1.Device1",
        "description": "",
        "browseName": "ThingWorxKepwareEdge#Device1",
        "writeMask": 0,
        "userWriteMask": 0,
        "typeDefinitionId": "i=61",
        "children": true
      }
    }

The Kepware server will fail to browse the returned node id if submitted for a subsequent browse (to find the nodes under the Device). There does not appear to be a configuration on the input parameter to `OpcTwinService.browse` (a `BrowseRequestApiModel`) that would allow the caller to specify the format of the result. 

**Expected behavior**
The node id returned by the browse call can be configured by the swagger client callee. 

**Screenshots**
See description

**Desktop (please complete the following information):**
- Edge modules are 2.5
- Deployment is 2.6, but was deployed from master mid-February

**Smartphone (please complete the following information):**
N/A

**Additional context**
Add any other context about the problem here.

cristipogacean, 2020-03-10:
would you be so kind and try this using 2.6:latest for both the edge modules (especially twin) and deployment? I just tried it on and works for me. 
btw, the expanded node Id is now uri formatted: E.g from my test:

browsed node :
 `"nodeid":"http://test.org/UA/Data/#i=10157"`
result: 
`{
  "node": {
    "nodeClass": "Object",
    "displayName": "Data",
    "nodeId": "http://test.org/UA/Data/#i=10157",
    "browseName": "http://test.org/UA/Data/#Data",
    "writeMask": 0,
    "userWriteMask": 0,
    "eventNotifier": "SubscribeToEvents",
    "children": true
  },
  "references": [
    {
      "referenceTypeId": "i=47",
      "direction": "Forward",
      "target": {
        "nodeClass": "Object",
        "displayName": "Static",
        "nodeId": "http://test.org/UA/Data/#i=10158",
        "browseName": "http://test.org/UA/Data/#Static",
        "writeMask": 0,
        "userWriteMask": 0,
        "eventNotifier": "SubscribeToEvents",
        "typeDefinitionId": "i=61",
        "children": true
      }
    }
  ]
}

pcameron-ptc, 2020-03-10:
We're having some other issues with 2.6 (see https://github.com/Azure/Industrial-IoT/issues/297), but I will try to see if I can get a browse to work. 

For what it's worth, we only have this problem (on the TKE server at least) for non-namespace zero nodes:

[image]

pcameron-ptc, 2020-03-10:
I tried it with 2.6 deployment and edge modules, and see the same result:

[image]

cristipogacean, 2020-03-11:
I tried to test in a simulated scenario similar like yours, but everything seem to work properly to me..
btw, the namespace uri is non-compliant in your server, that's ther reason you see the nsu in the expanden node ID. Anyhow I changed my server to simulate this situation and everything works fine in my swagger running from my browser.  
[image]
[image]

I kindof see a similar behavior from your printscreens, so can you please detail a bit what the real issue is?

pcameron-ptc, 2020-03-11:
This issue is identical to https://github.com/Azure/Industrial-IoT/issues/110. 

In other words, browsing any OPC UA server with a namespace URI entry that looks like "ThingworxKepwareEdge" through the OPC Twin will result in a browse response with a node id formatted in the "ExpandedNodeId" format due to the following code:  https://github.com/Azure/Industrial-IoT/blob/ea7da449fc62db31d6d6c0debdf2c75be462769f/components/opc-ua/src/Microsoft.Azure.IIoT.OpcUa.Protocol/src/Stack/Extensions/NodeIdEx.cs#L126

Regardless of whether or not this is a valid URI, it will cause the Industrial IoT stack to be incompatible with existing Kepware installations, because brownfield Kepware/TKE instances will fail browse requests submitted with node ids in the ExpandedNodeId format.

marcschier, 2020-03-11:
Hi @pcameron-ptc, The expanded node id is converted into a regular node id (ns + id) during the browse call against the Kepware server.    At the edge we use the uri from the nsu portion to look up the correct id in the namespace table, as the table is only valid per session, and the REST API is session less.   I assume the issue is unrelated to the node id format, and potentially somewhere else, i.e. what errors are you seeing for the subsequent call?   Could you share logs?

pcameron-ptc, 2020-03-11:
Thank you @marcschier and @cristipogacean , I wasn't aware of the nature of the fix and was surprised to still see the expanded node ids in the browse results of our 2.6 deployment. When I use the expanded node id as the input to the browse, the result is successful, as you have indicated. We can close this issue.

closed 2020-03-11 by pcameron-ptc