Azure/Industrial-IoT #306: OPC Twin with large number of nodes returns error whenever pagination should occur

caschw, 2020-03-11:
I have 10,000 generated references under one node. When I call browse on the OPC twin for that node, no continuation token is returned (null) and I get a BadNoConinuationPoints errorInfo object returned in the response

```
"errorInfo": {
    "diagnostics": [
      "BadNoContinuationPoints"
    ]
  }
```
What I expected was the first 100 nodes returned and a continuationToken that could be used with BrowseNext.

marcschier, 2020-03-11:
Hi @caschw, BadNoContinuationPoints is returned by the OPC UA server.  This error indicates that the server does not have enough simultaneously active continuation points and is running at limit.  If not otherwise specified in the request we set MaxReferencesToReturn to 0u, meaning the server should decide the batch sizes.   Maybe try to set it to MaxReferencesToReturn: 10000?.

pcameron-ptc, 2020-03-11:
@caschw , @marcschier is correct, it's up to the UA server to allocate some max number of continuation points. If, during the browse, the UA server has already used all of its allocated continuation points, then the browse will fail as you've indicated. 

You can observe the limit in any OPC UA server's Server.ServerCapabilities object:
[image]

marcschier, 2020-03-11:
Additional info - there seems to be only 100 items returned when MaxReferenceToReturn > 100 with above error.   Worth looking into the client or UA stack to see if this is an issue on the Twin side.

pcameron-ptc, 2020-03-12:
Update: Kepware does not read the MaxBrowseContinuationPoints from the object instance, but from the Type node. The max point count is hard-coded to 5 in the product. 

I'm continuing to investigate whether or not there is an issue in the Kepware Continuation Point browsing implementation:

[image]

pcameron-ptc, 2020-03-13:
@marcschier , I have some additional information regarding this issue. 

Carl's original description is correct - when we browse a node with 10k children using the Java swagger client (no value assigned to "withMaxReferencesToReturn"), we receive a continuation point and all 10k results (actually 10002 with a couple 'meta' data nodes thrown in). The continuation point in the response is null, and the error info is populated with no status code or error message, but with a diagnostic message that says "BadNoContinuationPoints". 

[image]

When we repeat the experiment using the swagger client, but set "withMaxReferencesToReturn" to 200, we receive 100 references in response. The continuation is non-null, as expected, but again the response is populated with "BadNoContinationPoints". 

[image]

In this case, when we submit the continuation point to BrowseNext, the call returns another 100 references. 

When I repeat this experiment using an OPC UA client (NetCoreConsoleClient from the OPCF stack), we see different results. In the case that no "MaxReferencesToReturn" is specified, we receive 1000 references and a valid continuation point. The service result is "Good". 

[image]

When we set the "MaxReferencesToReturn" to 200 and again use the NetCoreConsoleClient, we receive a valid continuation point and 200 references. Again, the service return is "Good". 

[image]

Can you confirm whether or not it is expected to see the following when using the swagger client?
1.) no 'statusCode' in the error info, and 2.) a max of 100 references returned, even when we specify a higher number.

marcschier, 2020-03-14:
Yes, the expectation is the rest call behaves the same as the reference client.  We will add a test case to our suite for above scenarios.

marcschier, 2020-03-17:
The main status code is from the actual browse call - the diagnostic info contains status of all service calls involved in the request, e.g. including value and attribute reads of reference nodes, as well as the browse call we use to get the "hasChildren" property.   Hence I am not surprised to see the additional diagnostics.   

I have added 3 unit tests and beefed up our test server to have a memory buffer with 10000 nodes.  I can read 5, 200, or just 1 and get exactly that number of nodes in the first call, without any error info set.   

I will look into whether the reference client does anything to "fill up" the max references to return value using multiple calls (e.g. getting 100, maxref-100 for the remainder).   Regardless, we should do that too, as the expectation from the REST call is pagination.  Working on it.

marcschier, 2020-03-17:
I looked through the stack code and could not find any special sauce.  I will try to reproduce this against KepServerEx now.   

What puzzles is the fact that browse runs out of continuation points, because we only request max=1 node to retrieve children, so no continuation point should be allocated as it is either 0 or 1.   To be super save I am now checking for a CP to be returned regardless and deleting it.

marcschier, 2020-03-17:
The first issue is identified - we had some legacy hold over code in the module that was triggered for any max > 100 to set it to 100 - this was a left over from when we did not have chunk server support and I removed it.

Still need to look into the CP issue with Kepware.

marcschier, 2020-03-25:
I couldn't identify anywhere where we leak the continuation.  Please regress with current preview build.

closed 2020-03-25 by marcschier