Azure/Industrial-IoT #307: ServiceResultApiModel data type does not match Swagger

caschw, 2020-03-11:
In [ServiceResultApiModel](https://github.com/Azure/Industrial-IoT/blob/master/modules/src/Microsoft.Azure.IIoT.Modules.OpcUa.Twin/src/v2/Models/ServiceResultApiModel.cs) line 52, status code is an unsigned int. In the [swagger document](https://github.com/Azure/Industrial-IoT/blob/master/api/swagger/twin.json) line 1043 it defines status code as int32. When using a tool to generate an SDK from the swagger documentation it does not use a data type that allows large enough status codes. 

I am using the AutoRest Code Generator to generate Java. It is creating the class with an Integer. If I manually change this to Long, then it works again.

marcschier, 2020-03-11:
We need to look at the swagger generation using swashbuckle to see how to surface a uint.

marcschier, 2020-03-16:
New swagger json in master, with integer format int64 for all uint type properties.

closed 2020-03-16 by marcschier