Azure/Industrial-IoT #309: 502 Error When Trying to Activate Endpoint

msha1026, 2020-03-13:
When trying to "Activate" and endpoint, sometimes a 502 response is returned with the following Response body:
```
<!DOCTYPE html PUBLIC "-//W3C//DTD XHTML 1.0 Strict//EN" "http://www.w3.org/TR/xhtml1/DTD/xhtml1-strict.dtd">
<html xmlns="http://www.w3.org/1999/xhtml">
<head>
    <meta http-equiv="Content-Type" content="text/html; charset=iso-8859-1" />
    <title>502 - Web server received an invalid response while acting as a gateway or proxy server.</title>
    <style type="text/css">
        < !-- body {
            margin: 0;
            font-size: .7em;
            font-family: Verdana, Arial, Helvetica, sans-serif;
            background: #EEEEEE;
        }
        fieldset {
            padding: 0 15px 10px 15px;
        }
        h1 {
            font-size: 2.4em;
            margin: 0;
            color: #FFF;
        }
        h2 {
            font-size: 1.7em;
            margin: 0;
            color: #CC0000;
        }
        h3 {
            font-size: 1.2em;
            margin: 10px 0 0 0;
            color: #000000;
        }
        #header {
            width: 96%;
            margin: 0 0 0 0;
            padding: 6px 2% 6px 2%;
            font-family: "trebuchet MS", Verdana, sans-serif;
            color: #FFF;
            background-color: #555555;
        }
        #content {
            margin: 0 0 0 2%;
            position: relative;
        }
        .content-container {
            background: #FFF;
            width: 96%;
            margin-top: 8px;
            padding: 10px;
            position: relative;
        }
        -->
    </style>
</head>
<body>
    <div id="header">
        <h1>Server Error</h1>
    </div>
    <div id="content">
        <div class="content-container">
            <fieldset>
                <h2>502 - Web server received an invalid response while acting as a gateway or proxy server.</h2>
                <h3>There is a problem with the page you are looking for, and it cannot be displayed. When the Web
                    server (while acting as a gateway or proxy) contacted the upstream content server, it received an
                    invalid response from the content server.</h3>
            </fieldset>
        </div>
    </div>
</body>
</html>
```

After restarting the twin module, this response does not happen again. We have seen this several times in the past week with the 2.6 deployment.

monikavar, 2020-03-18:
Hi @msha1026 , In order to investigate this issue, it would be super helpful to dig deeper into the logs. We have a tutorial on how to check logs from micro-services in Application Insights. Link is:
https://github.com/Azure/Industrial-IoT/blob/master/docs/tutorials/tut-applicationinsights.md

To check logs from webapp:
1. Please go to your app service and go to Deployment section and add username and password: 
[image]

2. Now go to Monitoring section and download logs from FTP (pls use full username as shown on this window):
[image]

Please provide us with the logs so we could know more about this issue. Thanks!

marcschier, 2020-03-29:
The HTML is returned by app service.  This appears to be an artifact of when the app service host reloads the service container instance which appears to happen from time to time.  We will move away from app service containers as soon as kudu supports .net core 3.1.  

Could you try this on a AKS deployment instead and see if this also happens?  We use a .net core 2 gateway there, and no containers.

msha1026, 2020-03-31:
Hi all, apologies for the late response. We haven't run into this issue since updating all of our edge modules to the latest tag version, so I will close this for now. Will reopen with logs if we come across this issue again. Thanks!

closed 2020-03-31 by msha1026