Azure/Industrial-IoT #319: Cannot add Kepware server application when it is remote from the IoT Edge Runtime

pcameron-ptc, 2020-03-16:
**Describe the bug**
When Kepware (TKS) is installed on a Windows 10 VM (same network / resource group as IoT Edge RT VM) then "add application" fails. 

**To Reproduce**
1. Deploy the IIoT 2.6 stack, including an edge with 2.6 modules
2. Install Kepware (TKS) on a Windows 10 VM in the same resource group
3. Update the Kepware configuration to expose a discovery URL to the network
4. Use Postman (or the Swagger client) to invoke AddApplication. 
5. The Kepware TKS server never appears in the list of applications. 

**Expected behavior**
The remote TKS server can be added

**Screenshots**

This illustrates a connection config using UA Expert:

[image]

This shows UA Expert connected using this configuration. UA Expert is running on my host in this example (the TKS server is running on an Azure hosted Windows 10 VM):

[image]

Here is the Kepware discovery URL configuration. Note the hostname is not fully qualified:

[image]

Here is Postman initiating the AddApplications call using "no security":

[image]

Finally, here is the discovery module output:

[image]

marcschier, 2020-03-17:
Hi @pcameron-ptc, would it be possible for you to send us the ARM template that you used to deploy the VM and IoT Edge VM or a description of the network configuration for both VM?   If the IoT Edge VM and cannot resolve the name or the resolved IP is not reachable then this might be happening.   We will see if we can replicate such a setup in our simulation environment.

cristipogacean, 2020-03-17:
@pcameron-ptc I managed to replicate the issue and we're working on a fix now. it seems the issue is that the dns resolves the hostname to an ipv6 address and the UA stack cannot properly cope with it. As a workaround til we fix it, you can try using the IPV4 address of the host in the discoveryUrl?

pcameron-ptc, 2020-03-19:
I'm still seeing this issue after verifying that the 2.6.116 version of the modules have been deployed to our IoT Edge Runtime. 

This screenshot captures the network configuration of the IoT Edge Runtime system:
[image]

This screenshot captures the network configuration of the Windows (TKS) system. Note that it shares a virtual network with the IoT Edge system:
[image]

Here is the Iot Edge system successfully connecting to port 49320 on the TKS system using telnet. Note that the DNS name "denali-vm" is resolved to the private IP 10.1.8.4
[image]

Here is the TKS system successfully pinging the IoT Edge system:
[image]

Here is the configuration of the "add-application" call in Postman
[image]

Finally, here are the results from tailing the discovery module logs:
[image]

cristipogacean, 2020-03-26:
the fix was pushed to master

closed 2020-03-26 by cristipogacean