Azure/Industrial-IoT #339: OPC Publisher: publish values that do not change?

apittier, 2020-03-20:
Hello, 

We're using the OPC Publisher edge module to subscribe to data changes on an OPC UA Server and send the corresponding telemetry messages to IoT Hub.

The thing is that we also need to send telemetry messages for OPC UA nodes values that do not change so often (e.g., configuration settings).

AFAIU, the OPC Publisher is able to send data **changes** but not "static/initial" values, is this correct? If so, is there a suggested approach to get these initial values sent to the IoT Hub?

Many thanks,
Amparo

cristipogacean, 2020-03-21:
In OPC UA when a client - in this case publisher creates a subscription and activates the monitored items, the server will push an initial data change message that contains the actual value of all monitored items regardless if the nodes to be monitored are "static/initial" or"dynamic". After this only the changes will be reported. Please let us know if you see a different behavior with your server.

jmbrunskill, 2020-03-25:
@cristipogacean Wouldn't this be a good scenario for the heartbeat functionality from 2.5.3? Is this implemented in 2.6 yet?

marcschier, 2020-03-25:
@jmbrunskill - fully aligned, that is the use case.

cristipogacean, 2020-04-03:
fixed in #362

closed 2020-04-03 by cristipogacean

apittier, 2020-04-03:
> In OPC UA when a client - in this case publisher creates a subscription and activates the monitored items, the server will push an initial data change message that contains the actual value of all [...]

Thanks for your answer. Yes, that's the behavior we observe with our opcuaserver.
I've been experimenting with the 'Heartbeat' interval as well, while creating subscriptions with more than 2500 nodes, and so far so good. 
BTW, we're using version: mcr.microsoft.com/iotedge/opc-publisher:2.5.1

jmbrunskill, 2020-04-06:
@apittier If you are using publisher 2.5.X you may want to look at this older ticket.
https://github.com/Azure/Industrial-IoT/issues/135

This issue is still in place for the 2.5.X publisher not sure if it will affect your use case but thought I'd share.

apittier, 2020-04-07:
> 
> 
> @apittier If you are using publisher 2.5.X you may want to look at this older ticket.
> #135
> 
> This issue is still in place for the 2.5.X publisher not sure if it will affect your use case but thought I'd share.

Thanks a lot @jmbrunskill ! I'll take a look at it.