Azure/Industrial-IoT #350: OPC Publisher: Hearbeat Not Working in v2.6.96

jerickcolendres, 2020-03-25:
Hi,

It seems the heartbeat functionality no longer works in  OPC Publisher v2.6.96. 

We configured a simulated tag **Sim.CH1.NZ010_SIM_CH1_TAG1\234754a-c63-b9601** in KepServerEx to toggle the value between 0 and 1 every minute. We then configured the OPC Publisher **pn.json** file with the following config and expect that OPC publisher will send a sample every 5 seconds:

```
 {
    "EndpointUrl": "opc.tcp://10.10.53.99:49320",
    "UseSecurity": false,
    "OpcNodes": [
      {
        "Id": "ns=2;s=Sim.CH1.NZ010_SIM_CH1_TAG1\\234754a-c63-b9601",
        "OpcSamplingInterval": 1000,
        "OpcPublishingInterval": 1000,
        "HeartbeatInterval": 5
      }
   ]
}
```
Here's the log from our IOT edge module that receives the samples from OPC Publisher (note that there's no sample every 5 seconds)
```
2020-03-25 04:14:16.802 [processor-thread-1] INFO  c.f.a.e.m.s.MessageProcessorService - Tracing [234754a-c63-b9601] received record:
 OpcPublisherDataModel(contentMask=1493, nodeId=nsu=ThingWorx Industrial Connectivity;s=Sim.CH1.NZ010_SIM_CH1_TAG1\234754a-c63-b9601, serverTimestamp=2020-03-25T04:14:16.407085Z, sourceTimestamp=2020-03-25T04:14:16.407085Z, statusCode=null, status=Good, applicationUri=urn:OPCUAHOST:PTC.ThingWorxIC.1:UA%20Server, timestamp=2020-03-25T04:14:16.7771598Z, value=OpcPublisherDataModel(type=11, value=1.0))
2020-03-25 04:15:16.797 [processor-thread-3] INFO  c.f.a.e.m.s.MessageProcessorService - Tracing [234754a-c63-b9601] received record:
 OpcPublisherDataModel(contentMask=1493, nodeId=nsu=ThingWorx Industrial Connectivity;s=Sim.CH1.NZ010_SIM_CH1_TAG1\234754a-c63-b9601, serverTimestamp=2020-03-25T04:15:16.4043086Z, sourceTimestamp=2020-03-25T04:15:16.4043086Z, statusCode=null, status=Good, applicationUri=urn:OPCUAHOST:PTC.ThingWorxIC.1:UA%20Server, timestamp=2020-03-25T04:15:16.7758741Z, value=OpcPublisherDataModel(type=11, value=0.0))
2020-03-25 04:16:16.792 [processor-thread-2] INFO  c.f.a.e.m.s.MessageProcessorService - Tracing [234754a-c63-b9601] received record:
 OpcPublisherDataModel(contentMask=1493, nodeId=nsu=ThingWorx Industrial Connectivity;s=Sim.CH1.NZ010_SIM_CH1_TAG1\234754a-c63-b9601, serverTimestamp=2020-03-25T04:16:16.4018699Z, sourceTimestamp=2020-03-25T04:16:16.4018699Z, statusCode=null, status=Good, applicationUri=urn:OPCUAHOST:PTC.ThingWorxIC.1:UA%20Server, timestamp=2020-03-25T04:16:16.7747373Z, value=OpcPublisherDataModel(type=11, value=1.0))
2020-03-25 04:17:16.792 [processor-thread-1] INFO  c.f.a.e.m.s.MessageProcessorService - Tracing [234754a-c63-b9601] received record:
 OpcPublisherDataModel(contentMask=1493, nodeId=nsu=ThingWorx Industrial Connectivity;s=Sim.CH1.NZ010_SIM_CH1_TAG1\234754a-c63-b9601, serverTimestamp=2020-03-25T04:17:16.3996991Z, sourceTimestamp=2020-03-25T04:17:16.3996991Z, statusCode=null, status=Good, applicationUri=urn:OPCUAHOST:PTC.ThingWorxIC.1:UA%20Server, timestamp=2020-03-25T04:17:16.7736457Z, value=OpcPublisherDataModel(type=11, value=0.0))
2020-03-25 04:18:16.792 [processor-thread-3] INFO  c.f.a.e.m.s.MessageProcessorService - Tracing [234754a-c63-b9601] received record:
 OpcPublisherDataModel(contentMask=1493, nodeId=nsu=ThingWorx Industrial Connectivity;s=Sim.CH1.NZ010_SIM_CH1_TAG1\234754a-c63-b9601, serverTimestamp=2020-03-25T04:18:16.3977373Z, sourceTimestamp=2020-03-25T04:18:16.3977373Z, statusCode=null, status=Good, applicationUri=urn:OPCUAHOST:PTC.ThingWorxIC.1:UA%20Server, timestamp=2020-03-25T04:18:16.7732723Z, value=OpcPublisherDataModel(type=11, value=1.0))
2020-03-25 04:19:16.793 [processor-thread-2] INFO  c.f.a.e.m.s.MessageProcessorService - Tracing [234754a-c63-b9601] received record:
 OpcPublisherDataModel(contentMask=1493, nodeId=nsu=ThingWorx Industrial Connectivity;s=Sim.CH1.NZ010_SIM_CH1_TAG1\234754a-c63-b9601, serverTimestamp=2020-03-25T04:19:16.3959894Z, sourceTimestamp=2020-03-25T04:19:16.3959894Z, statusCode=null, status=Good, applicationUri=urn:OPCUAHOST:PTC.ThingWorxIC.1:UA%20Server, timestamp=2020-03-25T04:19:16.7719076Z, value=OpcPublisherDataModel(type=11, value=0.0))
```

cristipogacean, 2020-04-03:
fixed in #362

closed 2020-04-03 by cristipogacean

omarine, 2020-05-26:
Hello,i juste tested the HeartbeatInterval from the master Branch but it seems not to be working 

anyone have tested it since ?

cristipogacean, 2020-05-26:
Hello @omarine, should be working in preview. What version did you use for testing?

omarine, 2020-05-26:
Hello @cristipogacean ,

My test was based on the master Branch
Should i base my tests on the lasted tagged version rather?

Thanks

cristipogacean, 2020-05-26:
No, please continue to use the :preview tag for the image. We would appreciate some details on your configuration.

omarine, 2020-05-26:
I actually run the opcpublisher locally ,because i have to add a feature for my client

But i can test the preview just in case.

omarine, 2020-05-26:
what is the latest stable branch i have to chose to base my modifications upon  please?

omarine, 2020-06-08:
Hello,

Any updates on the Heartbeat ,i am launching opcpublisher as a standalone from visualstudio, just tasted from the master branch but still not working 

my pn.json looks like below 

[image]

[image]

As an opcua server i have matrikon opc simulator exposing  DA endpoint,then i am using the Wrapper DA/UA

In case the heartbeat is working in preview,what branch is more inlined with the opcpublisher in the preview  ?

Thanks in advance

cristipogacean, 2020-06-08:
The preview is aligned to master branch. Heartbeat should work fine in the master.
Your publishing interval configured is a bit big in this case. Heartbeat interval should be multiple of publishing interval. Can you tune publishing interval to 1 sec or so?

omarine, 2020-06-08:
not working even with publishing interval to 1000 and heartbeatinterval to 5 sec,i have only changing values :(

I did the test with constant values as well,i should have the same values sent upstream with different timestamp ,but it's not the case

cristipogacean, 2020-06-08:
The heartbeat is basically replicating the last received datachange received from the server, so the value,, it's source timestamp and server timestamp are not being changed when heartbeat message is forced up. The publisher will not change these values since they are set by the server. The timestamp field in the message body will be nevertheless updated with the publisher's time. like in the example below, I just tried it out with the following arguments:
```
"--PkiRootPath=/mount/pki --pf=/mount/pn.json --fd=true --fm=true --bs=1 --di=20 --to --si=1"
```

and pn.json
```Json
[
	{
		"EndpointUrl": "opc.tcp://crpogace01:51210/UA/DemoServer",
		"UseSecurity": true,
		"OpcNodes": [
			{
				"Id": "http://test.org/UA/Data/#i=10217",
				"OpcSamplingInterval": 1000,
				"OpcPublishingInterval": 1000,
				"HeartbeatInterval": 5
			}
		]
	}
]
```
result in Azure IoT Explorer
```Json
5:32:48 PM, June 08, 2020:
{
  "body": {
    "NodeId": "http://test.org/UA/Data/#i=10217",
    "EndpointUrl": "opc.tcp://crpogace01:51210/UA/DemoServer",
    "ApplicationUri": "urn:CRPOGACE01:DemoServer",
    "Timestamp": "2020-06-08T15:32:48.0693087Z",
    "Value": {
      "Value": 15,
      "SourceTimestamp": "2020-06-08T08:54:32.7897422Z",
      "ServerTimestamp": "2020-06-08T15:32:32.1961491Z"
    },
    "SequenceNumber": 2,
    "ExtensionFields": {
      "PublisherId": "Standalone_CRPOGACE01_publisher",
      "DataSetWriterId": "opc.tcp://crpogace01:51210/UA/DemoServer_14a60b6dc4bda00e46019fd2f47c1e656deb4d6d"
    }
  },
  "enqueuedTime": "2020-06-08T15:32:48.272Z",
  "properties": {
    "$$ContentType": "application/x-monitored-item-json-v1",
    "iothub-message-schema": "application/opcua+uajson",
    "$$ContentEncoding": "utf-8"
  }
}
5:32:43 PM, June 08, 2020:
{
  "body": {
    "NodeId": "http://test.org/UA/Data/#i=10217",
    "EndpointUrl": "opc.tcp://crpogace01:51210/UA/DemoServer",
    "ApplicationUri": "urn:CRPOGACE01:DemoServer",
    "Timestamp": "2020-06-08T15:32:43.091219Z",
    "Value": {
      "Value": 15,
      "SourceTimestamp": "2020-06-08T08:54:32.7897422Z",
      "ServerTimestamp": "2020-06-08T15:32:32.1961491Z"
    },
    "SequenceNumber": 2,
    "ExtensionFields": {
      "PublisherId": "Standalone_CRPOGACE01_publisher",
      "DataSetWriterId": "opc.tcp://crpogace01:51210/UA/DemoServer_14a60b6dc4bda00e46019fd2f47c1e656deb4d6d"
    }
  },
  "enqueuedTime": "2020-06-08T15:32:43.379Z",
  "properties": {
    "$$ContentType": "application/x-monitored-item-json-v1",
    "iothub-message-schema": "application/opcua+uajson",
    "$$ContentEncoding": "utf-8"
  }
}
5:32:38 PM, June 08, 2020:
{
  "body": {
    "NodeId": "http://test.org/UA/Data/#i=10217",
    "EndpointUrl": "opc.tcp://crpogace01:51210/UA/DemoServer",
    "ApplicationUri": "urn:CRPOGACE01:DemoServer",
    "Timestamp": "2020-06-08T15:32:38.076743Z",
    "Value": {
      "Value": 15,
      "SourceTimestamp": "2020-06-08T08:54:32.7897422Z",
      "ServerTimestamp": "2020-06-08T15:32:32.1961491Z"
    },
    "SequenceNumber": 2,
    "ExtensionFields": {
      "PublisherId": "Standalone_CRPOGACE01_publisher",
      "DataSetWriterId": "opc.tcp://crpogace01:51210/UA/DemoServer_14a60b6dc4bda00e46019fd2f47c1e656deb4d6d"
    }
  },
  "enqueuedTime": "2020-06-08T15:32:38.300Z",
  "properties": {
    "$$ContentType": "application/x-monitored-item-json-v1",
    "iothub-message-schema": "application/opcua+uajson",
    "$$ContentEncoding": "utf-8"
  }
}
5:32:33 PM, June 08, 2020:
{
  "body": {
    "NodeId": "http://test.org/UA/Data/#i=10217",
    "EndpointUrl": "opc.tcp://crpogace01:51210/UA/DemoServer",
    "ApplicationUri": "urn:CRPOGACE01:DemoServer",
    "Timestamp": "2020-06-08T15:32:33.0803591Z",
    "Value": {
      "Value": 15,
      "SourceTimestamp": "2020-06-08T08:54:32.7897422Z",
      "ServerTimestamp": "2020-06-08T15:32:32.1961491Z"
    },
    "SequenceNumber": 1,
    "ExtensionFields": {
      "PublisherId": "Standalone_CRPOGACE01_publisher",
      "DataSetWriterId": "opc.tcp://crpogace01:51210/UA/DemoServer_14a60b6dc4bda00e46019fd2f47c1e656deb4d6d"
    }
  },
  "enqueuedTime": "2020-06-08T15:32:33.408Z",
  "properties": {
    "$$ContentType": "application/x-monitored-item-json-v1",
    "iothub-message-schema": "application/opcua+uajson",
    "$$ContentEncoding": "utf-8"
  }
}
```
when using --fm=false, then only the source timestamp is added to the message. In this case source timestamp will be overwritten with the publisher's actual time.

omarine, 2020-08-24:
Hello,

Do you have any idea when  the new version of the opcpublisher will be in GA ?

Regards,
Oussama