Azure/Industrial-IoT #359: Cannot add application by discovery id when only discoverer is online

pcameron-ptc, 2020-03-27:
**Describe the bug**
I have two IoT Edge VMs deployed. One is Linux/Ubuntu running TKE locally, the other is Windows 10 running TKS locally. In other words, on both machines, the OPC UA server is running on the same host as the discovery module. 

In this screenshot, we see that only the "TKS" vm is running:
[image]

Here is an attempt to add the TKS application using the TKS discoverer:
[image]

The edge indicates that it is able to find the endpoint and upload results:
[image]

However, no applications are ever returned:
[image]

When we power on the Linux VM, we get different results:
[image]

A second Add Applications request is required. After which, we see the TKS application has been added, but with the TKE discoverer:
[image]

The TKS endpoint can now be browsed:
[image]

The TKE endpoint can also now be browsed:
[image]

TKS is now successfully activated:
[image]

We're trying to understand if this is intended behavior. If so, what is the purpose of specifying the discover ID in the non-scanning explicit activation?

marcschier, 2020-03-29:
Hi @pcameron-ptc , the id in this case is a per request I'd that allows subscription and correlation of discovery progress and results.   There are substantial differences with what can be achieved on Windows container to host networking vs Linux.  Linux finds not only host but also other servers on the network.  Windows finds servers on the network but cannot access host servers.  That is why TKE is found by windows, and Linux finds TKS and TKE.  We are experimenting right now how we can work around this but we have no guarantees for a resolution.  My suggestion is to use Linux Host for IoT Edge only.

pcameron-ptc, 2020-03-30:
Thank you @marcschier - we have no particular requirement for Windows based IoT Edge runtime, and will follow your guidance to only use Linux Host for IoT Edge moving forward.

marcschier, 2020-05-04:
We won't fix for Windows.

closed 2020-05-04 by marcschier