Azure/Industrial-IoT #360: IIoT Edge UI Installer fails on first attempt on Windows 10 1809

pcameron-ptc, 2020-03-27:
**Describe the bug**
I have consistently seen this failure when running the Industrial IoT Gateway Installer for the first time on an Azure hosted Windows 10 1809 VM (D2_v3 size). It's not a blocker, a second run of the tool consistently completes successfully. 

[image]

marcschier, 2020-03-29:
After talking to my colleague this appears to be the result of the reboot prompt being suppressed on the VM in certain conditions.  The reboot is necessary after VC runtime installation.

barnstee, 2020-03-29:
Correct. I'll have a look to see if we can make this message a little more friendly.

barnstee, 2020-03-30:
This is now fixed:
[image]

barnstee, 2020-03-30:
New release on GitHub.

marcschier, 2020-03-30:
Hi @pcameron-ptc - please reopen if you still see issues with the installer.  Thanks!

closed 2020-03-30 by marcschier