Azure/Industrial-IoT #361: OPC Publisher direct method call from Azure Portal always reply with status 500

fgsantos89, 2020-03-28:
**Describe the bug**
OPC Publisher direct method call from Azure Portal always reply with {"status": 500, payload: null}

**To Reproduce**
Steps to reproduce the behavior:
1. Create the resources: Azure VM Ubuntu 18.04, Azure IoT Hub...
2. Follows the steps on https://docs.microsoft.com/en-us/azure/iot-edge/how-to-install-iot-edge-linux  to install the Azure Iot Edge Runtime / Hub
3. On Azure Iot Hub go to Edge Device click on Set Modules and Add Module OPC Publisher
4. Wait for the deploy
5. Go to Direct Method on the OPC Publisher Module and call method GetInfo with payload {} 

**Expected behavior**
Returning the info from the module

**Desktop:**
 - OS: Win 10
 - Browser Chrome  80.0.3987.149

`iotedge check`
`Configuration checks
--------------------
√ config.yaml is well-formed - OK
√ config.yaml has well-formed connection string - OK
√ container engine is installed and functional - OK
√ config.yaml has correct hostname - OK
√ config.yaml has correct URIs for daemon mgmt endpoint - OK
‼ latest security daemon - Warning
    Installed IoT Edge daemon has version 1.0.8 but 1.0.9 is the latest stable version available.
    Please see https://aka.ms/iotedge-update-runtime for update instructions.
√ host time is close to real time - OK
√ container time is close to host time - OK
‼ DNS server - Warning
    Container engine is not configured with DNS server setting, which may impact connectivity to IoT Hub.
    Please see https://aka.ms/iotedge-prod-checklist-dns for best practices.
    You can ignore this warning if you are setting DNS server per module in the Edge deployment.
‼ production readiness: certificates - Warning
    Device is using self-signed, automatically generated certs.
    Please see https://aka.ms/iotedge-prod-checklist-certs for best practices.
√ production readiness: certificates expiry - OK
√ production readiness: container engine - OK
‼ production readiness: logs policy - Warning
    Container engine is not configured to rotate module logs which may cause it run out of disk space.
    Please see https://aka.ms/iotedge-prod-checklist-logs for best practices.
    You can ignore this warning if you are setting log policy per module in the Edge deployment.

Connectivity checks
-------------------
√ host can connect to and perform TLS handshake with IoT Hub AMQP port - OK
√ host can connect to and perform TLS handshake with IoT Hub HTTPS / WebSockets port - OK
√ host can connect to and perform TLS handshake with IoT Hub MQTT port - OK
√ container on the default network can connect to IoT Hub AMQP port - OK
√ container on the default network can connect to IoT Hub HTTPS / WebSockets port - OK
√ container on the default network can connect to IoT Hub MQTT port - OK
√ container on the IoT Edge module network can connect to IoT Hub AMQP port - OK
√ container on the IoT Edge module network can connect to IoT Hub HTTPS / WebSockets port - OK
√ container on the IoT Edge module network can connect to IoT Hub MQTT port - OK
√ Edge Hub can bind to ports on host - OK

19 check(s) succeeded.
4 check(s) raised warnings. Re-run with --verbose for more details.`

`iotedge logs publisher`
Always showing:
`[20:25:06 INF] No job received, wait 00:00:05 ...(Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker)`

`iotedge logs edgeAgent`
`2020-03-28 20:23:39  Starting Edge Agent
2020-03-28 20:23:39.531 +00:00 Edge Agent Main()
<6> 2020-03-28 20:23:39.756 +00:00 [INF] - Initializing Edge Agent.
<6> 2020-03-28 20:23:39.950 +00:00 [INF] - Version - 1.0.9.29532150 (1f7cf6a21dfa365c8f21475553c544f4df02dd44)
[... 180 lines cut ...]
<6> 2020-03-28 20:24:31.571 +00:00 [INF] - Device connection removed for device opc-vm/publisher
<6> 2020-03-28 20:24:31.579 +00:00 [INF] - Remove device connection for device opc-vm/publisher
<6> 2020-03-28 20:24:31.622 +00:00 [INF] - Client opc-vm/publisher in device scope authenticated locally.
<6> 2020-03-28 20:24:31.622 +00:00 [INF] - Successfully generated identity for clientId opc-vm/publisher and username opc-vm/opc-vm/publisher/?api-version=2018-06-30&DeviceClientType=.NET%2F1.21.3%20%28.NET%20Core%203.1.2%3B%20Linux%205.0.0-1032-azure%20%2334-Ubuntu%20SMP%20Mon%20Feb%2010%2019%3A37%3A25%20UTC%202020%3B%20X64%29%20OpcPublisher_2.6.96.38252
<6> 2020-03-28 20:24:31.622 +00:00 [INF] - ClientAuthenticated, opc-vm/publisher, 5a087925
<6> 2020-03-28 20:24:31.624 +00:00 [INF] - New device connection for device opc-vm/publisher
<6> 2020-03-28 20:24:31.625 +00:00 [INF] - Bind device proxy for device opc-vm/publisher
<6> 2020-03-28 20:24:31.625 +00:00 [INF] - Binding message channel for device Id opc-vm/publisher
<6> 2020-03-28 20:24:31.625 +00:00 [INF] - Processing subscriptions TwinResponse, Methods, DesiredPropertyUpdates for client opc-vm/publisher.
<6> 2020-03-28 20:24:31.625 +00:00 [INF] - Processing pending subscriptions for opc-vm/publisher
<6> 2020-03-28 20:24:31.626 +00:00 [INF] - Set subscriptions from session state for opc-vm/publisher
<6> 2020-03-28 20:24:31.627 +00:00 [INF] - Processing subscriptions TwinResponse, Methods, DesiredPropertyUpdates for client opc-vm/publisher.
<6> 2020-03-28 20:24:31.628 +00:00 [INF] - Processing pending subscriptions for opc-vm/publisher
<6> 2020-03-28 20:24:31.628 +00:00 [INF] - Set subscriptions from session state for opc-vm/publisher
<6> 2020-03-28 20:24:31.682 +00:00 [INF] - Processing subscriptions TwinResponse, Methods, DesiredPropertyUpdates for client opc-vm/publisher.
<6> 2020-03-28 20:24:31.682 +00:00 [INF] - Processing pending subscriptions for opc-vm/publisher
<6> 2020-03-28 20:24:31.683 +00:00 [INF] - Set subscriptions from session state for opc-vm/publisher
<6> 2020-03-28 20:24:33.272 +00:00 [INF] - Updated reported properties for opc-vm/$edgeHub
<6> 2020-03-28 20:24:33.396 +00:00 [INF] - Updated reported properties for opc-vm/publisher
<6> 2020-03-28 20:29:24.588 +00:00 [INF] - Entering periodic task to reauthenticate connected clients
<6> 2020-03-28 20:34:24.581 +00:00 [INF] - Entering periodic task to reauthenticate connected clients
<6> 2020-03-28 20:39:24.581 +00:00 [INF] - Entering periodic task to reauthenticate connected clients
<6> 2020-03-28 20:44:24.584 +00:00 [INF] - Entering periodic task to reauthenticate connected clients`

I add the SimulatedTemperatureSensor edge Module and could call the direct method and also could call ping on the edgeAgent with status 200.

marcschier, 2020-03-29:
Hi @fgsantos89 , we deprecated and removed the direct method capabilities in the last release.  Please use the REST NodePublishStart and Stop API for publisher configuration.  Publishednodes.json configuration works as well.

closed 2020-03-29 by marcschier

fgsantos89, 2020-03-29:
Where I can find the documentation of the REST APIs ?

cristipogacean, 2020-03-29:
@fgsantos89 https://github.com/Azure/Industrial-IoT/tree/master/docs/api