Azure/Industrial-IoT #370: Publisher - Unable to publish nodes - Opc.Ua.ServiceResultException: BadRequestTimeout

JohnnyCrazy, 2020-04-03:
**Describe the bug**

I'm unable to publish nodes, following the guide [here](https://github.com/Azure/Industrial-IoT/blob/master/docs/tutorials/tut-use-cli.md)

I can successfully read the value (see CLI Output), but the publisher always fails with `BadRequestTimeout`.

<details>
<summary>CLI Output</summary>

```
PS C:\Users\t-jodell\projects\Industrial-IoT\api\src\Microsoft.Azure.IIoT.Api\cli> dotnet run console

  ___               _                 _            _           _           ___           _____
 |_ _|  _ __     __| |  _   _   ___  | |_   _ __  (_)   __ _  | |         |_ _|   ___   |_   _|
  | |  | '_ \   / _` | | | | | / __| | __| | '__| | |  / _` | | |  _____   | |   / _ \    | |
  | |  | | | | | (_| | | |_| | \__ \ | |_  | |    | | | (_| | | | |_____|  | |  | (_) |   | |
 |___| |_| |_|  \__,_|  \__,_| |___/  \__| |_|    |_|  \__,_| |_|         |___|  \___/    |_|

> endpoints list
==================
{
  "items": [
    {
      "registration": {
        "id": "uat743365211111972be2000a810a7fc9e22b514235",
        "endpointUrl": "opc.tcp://8ee3c451a270:50000/",
        "siteId": "jonas-moxa-box-2",
        "supervisorId": "jonas-moxa-box-2_module_twin",
        "discovererId": "jonas-moxa-box-2_module_discovery",
        "endpoint": {
          "url": "opc.tcp://172.17.0.3:50000/",
          "alternativeUrls": [
            "opc.tcp://172.17.0.1:50000/",
            "opc.tcp://8ee3c451a270:50000/",
            "opc.tcp://172.18.0.1:50000/",
            "opc.tcp://192.168.0.100:50000/"
          ],
          "securityMode": "None",
          "securityPolicy": "http://opcfoundation.org/UA/SecurityPolicy#None",
          "certificate": "MIIDezCCAmOgAwIBAgIIM/AOtWOgYyowDQYJKoZIhvcNAQELBQAwETEPMA0GA1UEAwwGT3BjUGxjMB4XDTIwMDQwMjA5MjUwNVoXDTIxMDQwMjA5MjUwNVowETEPMA0GA1UEAwwGT3BjUGxjMIIBIjANBgkqhkiG9w0BAQEFAAOCAQ8AMIIBCgKCAQEAq/7L6+tpbZ8Hlj5zmFP0vWBqjp9v+EYDfZCgOq0UnFhaqKBO0L7pg6sLWKlOrD7QF2MrgllV0fGomfFUr+mtjndFubU08+1LSuIMbXN3ZTwNTzcNaBN4fX0UnSnI0Wqpc9EgYUy0McmuQTWCPkcHp+s7EebEAUDojnYo6ooKsJMtRrPyBEO9LpXrAgxevN3DCGwrkZmhn3AW66xOxYHv0FJySyU83TmuE66thp9sRyepvtsTVO9xkc/fBrO6pHHhqeQSLKQrlOumYL+slO4kGVWiCof0OLweA4BGI2TigpHUA9tiw0ARRkRB/cEkamUm+UNEttlMXE9xv+5XnNahRwIDAQABo4HWMIHTMB0GA1UdDgQWBBSFewCT8FihA3Ryc9PS6WdbDAJtOTAMBgNVHRMBAf8EAjAAMEAGA1UdIwQ5MDeAFIV7AJPwWKEDdHJz09LpZ1sMAm05oRWkEzARMQ8wDQYDVQQDDAZPcGNQbGOCCDPwDrVjoGMqMA4GA1UdDwEB/wQEAwIC9DAgBgNVHSUBAf8EFjAUBggrBgEFBQcDAQYIKwYBBQUHAwIwMAYDVR0RBCkwJ4YXdXJuOk9wY1BsYzo4ZWUzYzQ1MWEyNzCCDDhlZTNjNDUxYTI3MDANBgkqhkiG9w0BAQsFAAOCAQEAk5Qvr+hRlMRsBGpe1PnmduqHah71mkMipwO7rcyn0DtKBij4PFRX/WJ0guk+XIFE28+MQMefpixhyWGF+BqQOXLqWY1BBKJTz10yBqV4OJiaFjiIFmTHY3a7u1VBMRni34LNbFCcJlbvxw/oAlWPFG9WF6fBfAyl3IaodDYiGEUXJyx+YrPoSMvU8p3UjDOAvXpI6AEXA6M+3zrzxkaEEZ9gDRZdDWlawneq+HkeVM8pYUY76koWMSBnPJIx8B/91QqMGnUXseVTs5cCWJ/ntGz72cPyVb/mOCaBARp+xeQQzmRI27j7zntIUzcz2Uxe3YB2u/b8rDL7tNdX+FnqfQ=="
        },
        "securityLevel": 0,
        "authenticationMethods": [
          {
            "id": "0",
            "credentialType": "None",
            "securityPolicy": "http://opcfoundation.org/UA/SecurityPolicy#Basic256"
          },
          {
            "id": "1",
            "credentialType": "UserName",
            "securityPolicy": "http://opcfoundation.org/UA/SecurityPolicy#Basic256"
          },
          {
            "id": "2",
            "credentialType": "X509Certificate",
            "securityPolicy": "http://opcfoundation.org/UA/SecurityPolicy#Basic256"
          }
        ]
      },
      "applicationId": "uas29e448cb17b7a07e3882bb74c7f25a1ef3160940",
      "activationState": "ActivatedAndConnected",
      "endpointState": "Ready"
    },
    {
      "registration": {
        "id": "uatc189a79f1ae5dc86de3b41f1e3e688db2d7f3178",
        "endpointUrl": "opc.tcp://8ee3c451a270:50000/",
        "siteId": "jonas-moxa-box-2",
        "supervisorId": "jonas-moxa-box-2_module_twin",
[... 114 lines cut ...]
[10:12:41 INF] New settings processed. (Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost)
[10:12:44 INF] No job received, wait 00:00:10 ... (Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker)
[10:12:46 INF] No job received, wait 00:00:10 ... (Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker)
[10:12:47 INF] No job received, wait 00:00:10 ... (Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker)
[10:12:47 INF] No job received, wait 00:00:10 ... (Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker)
[10:12:47 INF] No job received, wait 00:00:10 ... (Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker)
[10:12:54 INF] No job received, wait 00:00:10 ... (Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker)
[10:12:56 INF] No job received, wait 00:00:10 ... (Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker)
[10:12:57 INF] No job received, wait 00:00:10 ... (Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker)
```

</details>

**To Reproduce**
Steps to reproduce the behavior:
1. Follow the guide

**Expected behavior**
I expect that I receive a device to cloud message containing the value of the published node.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: `Linux Moxa 4.4.0-cip-uc8100a-me #12 Tue Dec 11 14:53:48 CST 2018 armv7l GNU/Linux`

cristipogacean, 2020-04-03:
It looks to me the host: 8ee3c451a270 from "endpointUrl": "opc.tcp://8ee3c451a270:50000/", is not accessible or not DNS resolved by the publisher module. Can you ping it ?

JohnnyCrazy, 2020-04-03:
Indeed, it's not reachable. I'm wondering how I can read the value from the CLI then? I would've guessed the CLI also gets its information from one of the docker containers on the device. Is it maybe because of the `NET_ADMIN` and `NetworkMode: Host` settings of the discovery container?

One of the solutions then is to hardcode the current LAN IP (`192.168.0.100`) when starting the opc-demo-server. Anyway, the opc-demo-server always takes the hostname as primary endpoint by default. The guide should be updated to reflect that?

marcschier, 2020-04-04:
There are alternate addresses on an endpoint model which the publisher needs to try as well.  That is because the endpoint url is returned by discovery but the discoverer found the endpoint under a different address. If the twin works this is the likely answer. Unless the publisher is run in standalone mode which it seems it is not.

marcschier, 2020-05-20:
The publisher now also attempty to use the alternative addresses identified by the Discovery module during discovery to establish a session.  If the twin can access the server, the publisher should now also.  Closing this issue, please re-open if publisher 2.7.75+ does not fix it.

closed 2020-05-20 by marcschier