Azure/Industrial-IoT #373: Twin is unresponsive - Unable to configure apps/publisher

JohnnyCrazy, 2020-04-06:
**Describe the bug**

The twin is not responding/useable after a restart of the demo opc-ua-server.

I had a working setup with the demo opc-ua-server running as a docker container on the IoT Edge Gateway after hard coding the hostname (see #370). I had to restart the demo server and saw the following errors in the twin console:

```
IssuerName: CN=OpcPlc creating session via opc.tcp://192.168.0.100:50000/ - retry.
[10:05:35 INF { name: "04266081-5f85-411e-9917-1d43b162e52a", sessionId: 2, url: "opc.tcp://192.168.0.100:50000/" }] Try to connect via opc.tcp://192.168.0.100:50000/ in 5000 ms...
[10:05:40 WRN { name: "04266081-5f85-411e-9917-1d43b162e52a", sessionId: 2, url: "opc.tcp://192.168.0.100:50000/" }] Establishing unencrypted connection.
[10:05:41 ERR Microsoft.Azure.IIoT.OpcUa.Protocol.Services.StackLogger] (Error) Certificate 'CN=OpcPlc' rejected. Reason=BadCertificateUntrusted
[10:05:41 ERR Microsoft.Azure.IIoT.OpcUa.Protocol.Services.StackLogger] (Error) Writing rejected certificate to directory: [Directory]pki/rejected
[10:05:41 INF { name: "04266081-5f85-411e-9917-1d43b162e52a", sessionId: 2, url: "opc.tcp://192.168.0.100:50000/" }] Certificate is not trusted.
SubjectName: CN=OpcPlc
```

I thought that makes sense, since the twin was not started with `AutoAccept=true`. I adapted the `twin` create options to the following and updated the module.

```
{
  "Cmd": [
    "--AutoAccept"
  ]
}
```

This didn't have any impact because it seems every change to the modules is overwritten after some seconds. Anyway, after the twin started again, I had the following output

```
[10:36:35 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Connecting all clients to moxa using MqttOverTcp.
[10:36:40 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.ClientServices] Application own certificate not found. Creating a new self-signed certificate with default settings...
[10:39:48 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.ClientServices] New application certificate with 304CC2450F538F68CA9870B27B38AC53B647B741, CN=Azure IIoT OPC Twin created
[10:39:48 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.ClientServices] Setting new application certificate 304CC2450F538F68CA9870B27B38AC53B647B741, CN=Azure IIoT OPC Twin...
[10:39:48 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.ClientServices] Adding own certificate to configured certificate store
[10:39:51 WRN Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Bypassing certificate validation for client.
[10:39:51 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Running in iotedge context.
[10:40:18 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] 0: Module jonas-moxa-box-2_twin reconnected due to Connection_Ok.
[10:40:23 INF Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] Initialize device twin for jonas-moxa-box-2 - twin
[10:40:26 INF Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] Applying initial desired state.
[10:40:26 INF Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] Reporting currently initial state.
[10:40:28 INF Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] Module Host started.
```

But my `apps` and `endpoints` were empty, so I tried to re-add them via the console:

```
apps add -u opc.tcp://192.168.0.100:50000 -a
```

which also triggered the discovery module:

```
[10:42:56 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] 24a39bf1-edbc-44a5-825c-28a2fbf031e4: Discovery operation started.
[10:42:56 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] 24a39bf1-edbc-44a5-825c-28a2fbf031e4: Searching 1 discovery urls for endpoints...
[10:42:56 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] 24a39bf1-edbc-44a5-825c-28a2fbf031e4: Trying to find endpoints on opc.tcp://192.168.0.100:50000/...
[10:42:57 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] 24a39bf1-edbc-44a5-825c-28a2fbf031e4: Found 1 endpoints on opc.tcp://192.168.0.100:50000/.
[10:42:57 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] 24a39bf1-edbc-44a5-825c-28a2fbf031e4: Found total of 1 servers ...
[10:42:57 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.DiscoveryServices] Uploading 1 results...
[10:42:57 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.DiscoveryServices] 1 results uploaded.
[10:42:57 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] 24a39bf1-edbc-44a5-825c-28a2fbf031e4: Discovery operation completed. 
```

But after that, nothing happens. The twin doesn't show any log and `apps` stays empty. Meanwhile, the publisher is working very good. It's uploading data every second and I'm unable to change/stop it, because the `endpoints` its using don't exist in the twin 😅

If you need more output of specific modules, happy to help.

**To Reproduce**
Steps to reproduce the behavior:

See above

**Expected behavior**

The twin should respond to `apps add -u opc.tcp://192.168.0.100:50000 -a`

JohnnyCrazy, 2020-04-06:
In an old log of `twin`, I see the following line:

```
[09:54:40 INF Microsoft.Azure.IIoT.OpcUa.Edge.Supervisor.Services.SupervisorServices] uat1fb652bb6c3de9bf68f5d3191ca93fb1b406887b twin started.
```

This is not printed in the current log for some reason. Seems like the supervisor of the twin never starts?

This is my `supervisors list`:

```
{
  "items": [
    {
      "id": "jonas-moxa-box-2_module_twin",
      "logLevel": "Information",
      "connected": true
    }
  ]
}
```

marcschier, 2020-04-06:
The twin needs to be activated. If you use the CLI you can run endpoint activate to do so for all endpoints or using the -i option just a single one.  

Gateways tagged as "iiotedge" are managed so that only managed workloads run on it (using layered deployments).

I will look into the rest.

JohnnyCrazy, 2020-04-07:
> The twin needs to be activated. If you use the CLI you can run endpoint activate to do so for all endpoints or using the -i option just a single one.

Yes, the problem I'm not able to activate the twin at all, since it's never listed in apps/endpoints:

```
> apps list
==================
{
  "items": []
}
==================
> endpoints list
==================
{
  "items": []
}
==================
> apps add -u opc.tcp://192.168.0.100:50000 -a
// Wait ~5 Minutes
> endpoints list
==================
{
  "items": []
}
==================
> apps list
==================
{
  "items": []
}
==================
```

The discovery shows the following:

```
[08:02:33 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] 2b5c6c62-b423-4f62-922a-ee84b3ba59bc: Discovery operation started.
[08:02:33 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] 2b5c6c62-b423-4f62-922a-ee84b3ba59bc: Searching 1 discovery urls for endpoints...
[08:02:33 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] 2b5c6c62-b423-4f62-922a-ee84b3ba59bc: Trying to find endpoints on opc.tcp://192.168.0.100:50000/...
[08:02:42 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] 2b5c6c62-b423-4f62-922a-ee84b3ba59bc: Found 1 endpoints on opc.tcp://192.168.0.100:50000/.
[08:02:43 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] 2b5c6c62-b423-4f62-922a-ee84b3ba59bc: Found total of 1 servers ...
[08:02:43 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.DiscoveryServices] Uploading 1 results...
[08:02:43 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.DiscoveryServices] 1 results uploaded.
[08:02:43 INF Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher] 2b5c6c62-b423-4f62-922a-ee84b3ba59bc: Discovery operation completed.
```

But all other logs are empty/of no interest. 

> Gateways tagged as "iiotedge" are managed so that only managed workloads run on it (using layered deployments).

Got it. So the only way to deploy `AutoAccept` would be to untag it and manually deploy the modules?

marcschier, 2020-05-19:
Yesterday I checked in a change to also allow "unmanaged" iot edge, i.e. an IoT Edge that is not fleet managed by the platform (and thus continuously re-updated).  Just set the "unmanaged": true tag in the gateway twin, but leave the __type__: "iiotedge" tag.   Then you can use az or portal to deploy modules manually to the gateway rather than through the automatic layered deployment approach.

marcschier, 2020-05-19:
The reason the connection from twin does not work anymore is that the opcplc changes its certificate every time it restarts.  The last discovered certificate (thumbprint) does thus not match the restarted plc.   AutoAccept is a dev only solution and a production security risk and this behavior is actually correct (e.g. imagine a rogue server impersonating using a different certificate).    

The reason the PLC changes its identity is due to the fact that when it is restarted the previous certificate is not persisted and thus a new one is created.  This is an issue in the simulation, less in practice where certificates are assigned for a longer duration.   

I created a seperate item to track publisher still publishing #473

Lets use this item to track the "apps add" not working for you.   

In that regard, is the Edge gateway tagged with `__type__: "iiotedge"` could you check?   Because the onboarding service (the one that creates the identity in IoTHub from the discovery result contextualizes the identity with the supervisor in the same gateway, and if it cannot find the gateway or a supervisor in it in the registry (using the tag above as a query) it will discard the result.

JohnnyCrazy, 2020-05-19:
> In that regard, is the Edge gateway tagged with __type__: "iiotedge" could you check?

Yes, it was tagged as edge gateway. I sadly had to redo my local iiot-setup so I'm unable to do any testing in this case.

Could the change of the PLC certificate affect `apps add`?

> The reason the connection from twin does not work anymore is that the opcplc changes its certificate every time it restarts. The last discovered certificate (thumbprint) does thus not match the [...]

Got it. Since my/our primary goal is to create a customer hands-on workshop for an IIoT Setup, the dev only solution is our go-to currently. With the knowledge that layered deployment are being used, we know where to look now tho! :)

marcschier, 2020-05-20:
I have tested apps add -u on master branch (2.7.76) and was able to register and remove ok.   The issue seen here seems to be related to the setup of the gateway is my hunch as the discovery provided a single result and uploaded it, the processing of it failed however as no application is showing up.   The latter issue has been due to the gateway running the discovery workload not being registered as `__type__: iiotedge` or no opc twin module deployed to said gateway.   As the core "add -u" functionality works, I am moving this for further investigation into 2.8.

closed 2021-02-26 by marcschier