Azure/Industrial-IoT #380: Failure to deserialize by swagger client when reading a array nodes

pcameron-ptc, 2020-04-13:
**Describe the bug**
The opcTwinService.readValueAsync swagger client method (java generated code) fails with the following message when reading a node of type "double array" (similar error for other array types):

"com.fasterxml.jackson.databind.exc.MismatchedInputException: Cannot deserialize instance of `java.lang.String` out of START_ARRAY token
 at [Source: (String)"{
  "value": [
    12.34,
    12.34,
    12.34,
    12.34
  ],
  "dataType": "Double",
  "sourceTimestamp": "2020-04-13T20:45:06.182Z",
  "serverTimestamp": "2020-04-13T20:45:06.182Z"
}"; line: 11, column: 1] (through reference chain: com.microsoft.azure.iiot.ua.sdk.swagger.twin.models.ValueReadResponseApiModel["value"])" 

The read succeeds when performed by Postman:
[image]

UA Expert is able to monitor the node:
[image]

marcschier, 2020-04-14:
My assumption is that the swagger.json sets the variantvalue object as string intead of object.

marcschier, 2020-04-14:
Yes - confirmed -
       "value": {
          "description": "Value read",
          "type": "string"
        },

Will be fixed.

pcameron-ptc, 2020-04-14:
Thanks Marc!

marcschier, 2020-05-04:
Fix is in master.

closed 2020-05-04 by marcschier