Azure/Industrial-IoT #385: How to authenticate with a Certificate with Twin?

andr5362, 2020-04-15:
I am trying to read a node/value or write a node/value. I can get it to work without the certificate validation on both sides. But when I add the certificates I can't get it to work.

How do I add a certificate to be trusted/expected instead of rejected? And how do I set up a client certificate on the twin at should be trusted on the server-side in my case a PLC? (Not the auto-generated one, but if I would choose that how can I access it through the X509 Certificate store on Windows 10?)

`{
    "endpoint": {
        "url": "opc.tcp://10.30.98.251:4840",
        "securityMode": "SignAndEncrypt",
        "certificate": "Thumbprint for server or client?"
    },
    "request": {
        "header": {
            "elevation" : {
                "type": "X509Certificate",
                "value": ".cert or .pfx for server or client?"
            }
        },
        "NodeId": "ns=3;s=\"OPC-UA_COMM\".\"Test\".\"Test\"" 
    }
}`

andr5362, 2020-04-16:
One of the errors i get if i parse the .cer in the evaluation is following

```
[DirectMethod] Invoking Direct Method [ValueRead_V2] to [DKDN01AS1133/opctwin] ...
[DirectMethod] Response from [DKDN01AS1133/opctwin]:
{
  "status": 401,
  "payload": {
    "ClassName": "System.UnauthorizedAccessException",
    "Message": null,
    "Data": null,
    "InnerException": {},
    "HelpURL": null,
    "StackTraceString": "   at Microsoft.Azure.IIoT.Modules.OpcUa.Twin.Controllers.SupervisorMethodsController.ValueReadAsync(EndpointApiModel endpoint, ValueReadRequestApiModel request) in D:\\a\\1\\s\\modules\\src\\Microsoft.Azure.IIoT.Modules.OpcUa.Twin\\src\\Controllers\\SupervisorMethodsController.cs:line 141",
    "RemoteStackTraceString": null,
    "RemoteStackIndex": 0,
    "ExceptionMethod": null,
    "HResult": -2147024891,
    "Source": "System.Private.CoreLib",
    "WatsonBuckets": null,
    "Exception": "UnauthorizedAccessException",
    "CausedBy": {
      "Message": "Endpoint does not support the user identity type provided.",
      "Exception": "ServiceResultException"
    }
  }
}
```

marcschier, 2020-04-16:
This is not yet supported but is in our backlog.   We will evaluate user certificates for a release later this year.

andr5362, 2020-04-17:
Okay, what are the other options? Can it forward its own generated certificate to the OPC Server?

marcschier, 2020-04-21:
Hi @andr5362 - the certificate property is a thumbprint of a server certficiate on the endpoint you want to *trust*.   the certificate in the user elevation is a user authentication certificate.   This is being deprecated in favor of tokens, but we are looking at providing both token and certificate user authentication at some point in the future potentially.   Can you use username/password or anonymous authentication over encrypted channel as alternative?

hansgschossmann, 2020-10-02:
@andr5362 for now the workaround would be to change the configuration for the OPC Twin modules layered deployment configuration to bind the certificate directory to a host directory. Please let me know if you still need to resolve this.

marcschier, 2023-06-25:
Will be tracked in #2005.

closed 2023-06-25 by marcschier