Azure/Industrial-IoT #386: The deploy.sh script fails with Conflict error message for the logs module

szmgabor, 2020-04-16:
**The bug**
I've been trying to deploy the Azure Industrial IoT Platform from this [guide](https://azure.github.io/Industrial-IoT/deploy/howto-deploy-all-in-one.html). 
The deployment always fails with the following error message:

```
7:55:17 AM - Resource Microsoft.OperationalInsights/workspaces 'logs-unnxae3g' failed with message '{
  "error": {
    "code": "Conflict",
    "message": "The workspace name 'logs-unnxae3g' is not unique",
    "target": "name"
  }
}'
Deployment failed.
```

**To Reproduce**
Steps to reproduce the behavior:
- Follow the steps described in this [guide](https://azure.github.io/Industrial-IoT/deploy/howto-deploy-all-in-one.html).
- During the deployment, after the resource group and the application name were specified I get the error message mentioned above.

**Expected behavior**
I would expect the deployment to be successful. In the Azure Portal under the newly created Resource Group I can see that there are several resources/services allocated.

**Environment:**
I'm running the deployment script on a VirtualBox Virtual Machine with Ubuntu 18.04.4 LTS (Bionic Beaver).

marcschier, 2020-04-16:
Unfortunately the workspace name must be worldwide unique.   We will increase the randomness in the name to make conflicts less likely.

szmgabor, 2020-04-17:
Changing the Resource Group name worked. Thanks for the help.

closed 2020-04-17 by szmgabor