Azure/Industrial-IoT #388: Can't browse from the Azure IIoT Platform 2.7.11

szmgabor, 2020-04-20:
**Describe the bug**
I have deployed the Azure Industrial IoT Platform as described [here](https://github.com/Azure/Industrial-IoT/blob/master/docs/deploy/howto-deploy-all-in-one.md).
Afterwards I set up my onw Industrial IoT Gateway as described [here](https://github.com/Azure/Industrial-IoT/blob/master/docs/deploy/quickstart-gateway-installer.md) on a Ubuntu 18.04.3 LTS.
When the GW was up and running I was able to see it in the Engineering tool under the Discovery menu. I configured the discovery and found the OPC UA Server to which I wanted to connect.
From the Browse menu I connected to the correct endpoint.
But when I click on the name of the endpoint nothing happens.
I checked the logs of the front-end and I can see the following message:

```
[2020-04-20T13:44:59.977Z] Information: Normalizing '_blazor' to 'https://softing-edge-iiot.azurewebsites.net/_blazor'.	blazor.server.js:1 
[2020-04-20T13:45:01.592Z] Information: WebSocket connected to wss://softing-edge-iiot.azurewebsites.net/_blazor?id=o6NIOwh7EfokwkRLO9h4ag. blazor.server.js:15 
[2020-04-20T13:45:13.731Z] Error: There was an unhandled exception on the current circuit, so this circuit will be terminated. For more details turn on detailed exceptions by setting 'DetailedErrors: true' in 'appSettings.Development.json' or set 'CircuitOptions.DetailedErrors'. 
	e.log @ blazor.server.js:15
	C @ blazor.server.js:8
	(anonymous) @ blazor.server.js:8
	(anonymous) @ blazor.server.js:1
	e.invokeClientMethod @ blazor.server.js:1
	e.processIncomingData @ blazor.server.js:1
	connection.onreceive @ blazor.server.js:1
	i.onmessage @ blazor.server.js:1
	blazor.server.js:1 
[2020-04-20T13:45:13.732Z] Information: Connection disconnected.
```

**To Reproduce**
Steps to reproduce the behavior:
1. Start an IIoT Gateway
2. Discover the servers from the Engineering Tool
3. Connect to the correct endpoint
4. Try to browse the connected endpoint. 

**Expected behavior**
I would expect to be able to browse the endpoint.

**Additional context**
After this problem occures I can't interact with the webpage. I can switch between the menus but other than the loading spinner no content is displayed. It seems that the websocket connection is closed and it's not opened again. The page seems to recover after a Hard Reload, but the browsing still doesn't work.
There doesn't seem to be any error in either of the edge modules (discovery, opctwin, opcpublisher) I'm assuming this is a webapp/frontend issue.

szmgabor, 2020-04-21:
In some cases after the failed browse attempt, when reloading the page a Internal Server Error page appears, like the one below:
[image]

The exact message of the error is this:

```
System.NullReferenceException: Object reference not set to an instance of an object.
   at Microsoft.Azure.IIoT.App.Pages.Browser.BuildRenderTree(RenderTreeBuilder __builder)
   at Microsoft.AspNetCore.Components.Rendering.ComponentState.RenderIntoBatch(RenderBatchBuilder batchBuilder, RenderFragment renderFragment)
   at Microsoft.AspNetCore.Components.RenderTree.Renderer.RenderInExistingBatch(RenderQueueEntry renderQueueEntry)
   at Microsoft.AspNetCore.Components.RenderTree.Renderer.ProcessRenderQueue()
--- End of stack trace from previous location where exception was thrown ---
   at Microsoft.AspNetCore.Components.Rendering.HtmlRenderer.HandleException(Exception exception)
   at Microsoft.AspNetCore.Components.RenderTree.Renderer.ProcessRenderQueue()
   at Microsoft.AspNetCore.Components.RenderTree.Renderer.AddToRenderQueue(Int32 componentId, RenderFragment renderFragment)
   at Microsoft.AspNetCore.Components.ComponentBase.StateHasChanged()
   at Microsoft.AspNetCore.Components.ComponentBase.CallOnParametersSetAsync()
   at Microsoft.AspNetCore.Components.ComponentBase.RunInitAndSetParametersAsync()
   at Microsoft.AspNetCore.Components.Rendering.HtmlRenderer.HandleException(Exception exception)
   at Microsoft.AspNetCore.Components.Rendering.ComponentState.SetDirectParameters(ParameterView parameters)
   at Microsoft.AspNetCore.Components.RenderTree.Renderer.RenderRootComponentAsync(Int32 componentId, ParameterView initialParameters)
   at Microsoft.AspNetCore.Components.Rendering.HtmlRenderer.CreateInitialRenderAsync(Type componentType, ParameterView initialParameters)
   at Microsoft.AspNetCore.Components.Rendering.HtmlRenderer.RenderComponentAsync(Type componentType, ParameterView initialParameters)
   at Microsoft.AspNetCore.Components.Rendering.RendererSynchronizationContext.<>c__11`1.<<InvokeAsync>b__11_0>d.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at Microsoft.AspNetCore.Mvc.ViewFeatures.StaticComponentRenderer.PrerenderComponentAsync(ParameterView parameters, HttpContext httpContext, Type componentType)
   at Microsoft.AspNetCore.Mvc.ViewFeatures.ComponentRenderer.PrerenderedServerComponentAsync(HttpContext context, ServerComponentInvocationSequence invocationId, Type type, ParameterView parametersCollection)
   at Microsoft.AspNetCore.Mvc.ViewFeatures.ComponentRenderer.RenderComponentAsync(ViewContext viewContext, Type componentType, RenderMode renderMode, Object parameters)
   at Microsoft.Azure.IIoT.App.Pages.Pages__Host.<ExecuteAsync>b__12_1() in D:\a\1\s\samples\src\Microsoft.Azure.IIoT.App\src\Pages\_Host.cshtml:line 19
   at Microsoft.AspNetCore.Razor.Runtime.TagHelpers.TagHelperExecutionContext.SetOutputContentAsync()
   at Microsoft.Azure.IIoT.App.Pages.Pages__Host.ExecuteAsync()
   at Microsoft.AspNetCore.Mvc.Razor.RazorView.RenderPageCoreAsync(IRazorPage page, ViewContext context)
   at Microsoft.AspNetCore.Mvc.Razor.RazorView.RenderPageAsync(IRazorPage page, ViewContext context, Boolean invokeViewStarts)
   at Microsoft.AspNetCore.Mvc.Razor.RazorView.RenderAsync(ViewContext context)
   at Microsoft.AspNetCore.Mvc.ViewFeatures.ViewExecutor.ExecuteAsync(ViewContext viewContext, String contentType, Nullable`1 statusCode)
   at Microsoft.AspNetCore.Mvc.ViewFeatures.ViewExecutor.ExecuteAsync(ViewContext viewContext, String contentType, Nullable`1 statusCode)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ResourceInvoker.<InvokeNextResultFilterAsync>g__Awaited|29_0[TFilter,TFilterAsync](ResourceInvoker invoker, Task lastTask, State next, Scope scope, Object state, Boolean isCompleted)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ResourceInvoker.Rethrow(ResultExecutedContextSealed context)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ResourceInvoker.ResultNext[TFilter,TFilterAsync](State& next, Scope& scope, Object& state, Boolean& isCompleted)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ResourceInvoker.InvokeResultFilters()
--- End of stack trace from previous location where exception was thrown ---
   at Microsoft.AspNetCore.Mvc.Infrastructure.ResourceInvoker.<InvokeNextResourceFilter>g__Awaited|24_0(ResourceInvoker invoker, Task lastTask, State next, Scope scope, Object state, Boolean isCompleted)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ResourceInvoker.Rethrow(ResourceExecutedContextSealed context)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ResourceInvoker.Next(State& next, Scope& scope, Object& state, Boolean& isCompleted)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ResourceInvoker.InvokeFilterPipelineAsync()
--- End of stack trace from previous location where exception was thrown ---
   at Microsoft.AspNetCore.Mvc.Infrastructure.ResourceInvoker.<InvokeAsync>g__Logged|17_1(ResourceInvoker invoker)
   at Microsoft.AspNetCore.Routing.EndpointMiddleware.<Invoke>g__AwaitRequestTask|6_0(Endpoint endpoint, Task requestTask, ILogger logger)
   at Microsoft.AspNetCore.Authorization.AuthorizationMiddleware.Invoke(HttpContext context)
   at Microsoft.AspNetCore.Authentication.AuthenticationMiddleware.Invoke(HttpContext context)
   at Microsoft.AspNetCore.Diagnostics.DeveloperExceptionPageMiddleware.Invoke(HttpContext context)
```

In which service should I try to look into, what could be the source of the problem?

dacolgit, 2020-04-29:
which server are you using?
From the debug tools of your browser could you send the console output?

szmgabor, 2020-05-07:
This is the debug log from the browser:

```
[2020-05-07T08:08:01.444Z] Error: There was an unhandled exception on the current circuit, so this circuit will be terminated. For more details turn on detailed exceptions by setting 'DetailedErrors: true' in 'appSettings.Development.json' or set 'CircuitOptions.DetailedErrors'.		 blazor.server.js:15 
	e.log	@	blazor.server.js:15
[2020-05-07T08:08:01.449Z] Information: Connection disconnected.			blazor.server.js:1 
```

And I get the following in the Log Stream from the corresponding app service:

```
2020-05-07T08:07:48.406046538Z [08:07:48 INF] Authorization was successful.
2020-05-07T08:07:48.406191439Z [08:07:48 INF] Authorization was successful.
2020-05-07T08:07:48.406321840Z [08:07:48 INF] Authorization was successful.
2020-05-07T08:07:48.407026744Z [08:07:48 WRN] Unhandled exception rendering component: Object reference not set to an instance of an object.
2020-05-07T08:07:48.407044044Z System.NullReferenceException: Object reference not set to an instance of an object.
2020-05-07T08:07:48.407050544Z    at Microsoft.Azure.IIoT.App.Pages.Browser.BuildRenderTree(RenderTreeBuilder __builder)
2020-05-07T08:07:48.407064645Z    at Microsoft.AspNetCore.Components.Rendering.ComponentState.RenderIntoBatch(RenderBatchBuilder batchBuilder, RenderFragment renderFragment)
2020-05-07T08:07:48.407069645Z    at Microsoft.AspNetCore.Components.RenderTree.Renderer.RenderInExistingBatch(RenderQueueEntry renderQueueEntry)
2020-05-07T08:07:48.407074145Z    at Microsoft.AspNetCore.Components.RenderTree.Renderer.ProcessRenderQueue()
2020-05-07T08:07:48.407627748Z [08:07:48 ERR] Unhandled exception in circuit 'aZ-MEEPNVp5b4hmQxNXG51Ya6IMJpI6-pjdTOIwjWOU'.
2020-05-07T08:07:48.407641448Z System.NullReferenceException: Object reference not set to an instance of an object.
2020-05-07T08:07:48.407646748Z    at Microsoft.Azure.IIoT.App.Pages.Browser.BuildRenderTree(RenderTreeBuilder __builder)
2020-05-07T08:07:48.407651248Z    at Microsoft.AspNetCore.Components.Rendering.ComponentState.RenderIntoBatch(RenderBatchBuilder batchBuilder, RenderFragment renderFragment)
2020-05-07T08:07:48.407655648Z    at Microsoft.AspNetCore.Components.RenderTree.Renderer.RenderInExistingBatch(RenderQueueEntry renderQueueEntry)
2020-05-07T08:07:48.407660048Z    at Microsoft.AspNetCore.Components.RenderTree.Renderer.ProcessRenderQueue()
```

dacolgit, 2020-05-19:
Closed as not reproducible with latest preview