Azure/Industrial-IoT #389: Question: How to make OPC Publisher add the OPC UA StatusCode erros  such as Bad Communication in its JSON Payloads?

ssvillamor23, 2020-04-21:
I have an issue where some of the PLC (with OPCUA-Server)  had been turned off makin the OPC Publisher has this error Bad Communication in its logs. I wanted to get these errors so I could monitor the reason what is the cause why there had been no updates of telemetry at a certain period of time. Is there anyway to resolve by any means like configuring the OPC Publisher? or Making another IOT Edge Module for it , if yes, where do I see those documentations?
Thanks for your Responses?

jmbrunskill, 2020-04-21:
@ssvillamor23 - Which version of OPC Publisher are you using? There have been some changes to these formats in recent versions.

With version 2.5.X you can use a telemetry.json file to add opcua status to the telemetry stream.

This can be configured with a module startup parameter like this:
`--tc=/appdata/telemetry.json`

Our telemetry.json looks something like this:
```
 {
        "Defaults": {
            "EndpointUrl": {
                "Publish": false,
                "Pattern": "(.*)",
                "Name": "EndpointUrl"

            },
            "NodeId": {
                "Publish": true,
                "Name": "NodeId"
            },
            "MonitoredItem": {
                "Flat": true,

                "ApplicationUri": {
                    "Publish": true,
                    "Name": "ApplicationUri"
                },
                "DisplayName": {
                    "Publish": true,
                    "Name": "DisplayName"
                }
            },
            "Value": {
                "Flat": false,

                "Value": {
                    "Publish": true,
                    "Name": "Value"
                },
                "SourceTimestamp": {
                    "Publish": true,
                    "Name": "SourceTimestamp"
                },
                "StatusCode": {
                    "Publish": true,
                    "Name": "StatusCode"
                },
                "Status": {
                    "Publish": false,
                    "Name": "Status"
                }
            }
        }
    }
```
See: https://github.com/Azure/Industrial-IoT/blob/2.5.3/docs/modules/publisher.md#telemetry-formats

For version 2.6 onwards there has been some confusion around this for us.
We are waiting on confirmation, but I believe the IIOT team is going to establish a backwards compatible telemetry format which includes the OPC UA status in it when using 'legacy mode' configuration (e.g. published_nodes.json)

With 2.6 there is also a pub-sub model which has a format more closely aligned to the OPC UA standard, and includes a binary protocol. Currently this can only be configured via the cloud.

See Also:
https://github.com/Azure/Industrial-IoT/blob/master/docs/modules/publisher.md#breaking-changes-in-26x-from-25x-and-lower-versions

I'm hoping the documentation is updated to clarify some of this soon.

ssvillamor23, 2020-04-27:
HI

> @ssvillamor23 - Which version of OPC Publisher are you using? There have been some changes to these formats in recent versions.
>
> With version 2.5.X you can use a telemetry.json file to add opcua [...]

Thanks you so much  for your suggestion @jmbrunskill. 
I did what you suggests but I just recently notice that  OPC Publisher suppresses errors such as BadNoCommunication and BadWaitingForInitialData, making the OPC Publisher does not sent any messages at all.
In order to disable this, I simply just just add --ss=BadWaitingForInitialData and didn't add BadNoCommunication so I could expect it would send any messages relating to connection issue with the OPC UA Server. But the problem is, the errors comes out in the OPCPublisher log and not in the telemetry payload.

-- heartbeat doesnt help at all, since it just send the last value of the message for the node it is intended to.

Should there be any way that the OPC Publisher should sent a published message at a certain period, but the telemetry valie would be something like INVALID or NULL and has a OPCUA status code so I could know what makes it fail?

cristipogacean, 2020-05-12:
in 2.7.x a message with a bad status code valuechange looks like this:

```
{
  "body": {
    "NodeId": "http://opcfoundation.org/Quickstarts/ReferenceApplications#s=%2fStatic%2fAll+Profiles%2fScalar%2fAccessLevel_CurrentRead_NotUser",
    "ApplicationUri": "urn:IndustrialHost01:OpcUaServer",
    "Value": {
      "StatusCode": {
        "Symbol": "BadUserAccessDenied",
        "Code": 2149515264
      },
      "SourceTimestamp": "2020-04-28T11:41:31.7725401Z"
    }
  },
  "enqueuedTime": "2020-04-28T11:41:32.896Z",
  "properties": {
    "iothub-creation-time-utc": "2020-04-28T11:41:32.5830272Z",
    "$$ContentType": "application/x-monitored-item-json-v1",
    "iothub-message-schema": "application/opcua+uajson",
    "$$ContentEncoding": "utf-8",
    "$$CreationTimeUtc": "04/28/20 11:41:32 AM"
  }
}
```
the `--ss` is no longer supported in 2.6+ and there is no logic on suppressing valuechanges with Bad statuses.

barnstee, 2020-09-17:
Closing due to inactivity. Question has been answered.

closed 2020-09-17 by barnstee