Azure/Industrial-IoT #392: Discovery cannot be triggered in 2.6.146

vslepakov, 2020-04-22:
**Describe the bug**
Cannot trigger discovery. It fails with an error in the browser console

**To Reproduce**
Steps to reproduce the behavior:
1. In the engineering tool Go to '/discoverers'
2. Wait for the list to appear
3. On a connected discovery module click the checkbox 'Enable Scanning.'
4. In the Browser console you see an error and discovery is not executed

It appears that the web socket connection is already closed when trying to enable scanning.
It is also impossible to change the scanning settings. 

**Expected behavior**
Discovery works

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows
 - Browser: Chrome
 - Version: 80.0.3987.163 (Official Build) (64-bit)

dacolgit, 2020-04-23:
Take a look [here](https://github.com/Azure/Industrial-IoT/blob/master/docs/deploy/upgrade.md)

vslepakov, 2020-04-23:
yeah, **Default** is already configured. still same problem.

dacolgit, 2020-04-24:
You are using a 2.6 code with a 2.7 deployment. 
For 2.6 you need ServiceMode = Serverless in SignalR
For 2.7 you need ServiceMode = Default in SignalR

vslepakov, 2020-04-24:
Ok, switching to **Serverless** did help, thanks!
We actually just deployed using the deployment script. Why is this settings not automatically applied correctly?

dacolgit, 2020-04-24:
Do you follow instructions [here](https://github.com/Azure/Industrial-IoT/blob/master/docs/deploy/howto-deploy-all-in-one.md) using deploy script from root folder?

vslepakov, 2020-04-24:
A colleague of mine used the all-in-one option. so we don't have a k8s cluster, it is all app services

vslepakov, 2020-04-24:
I will go ahead and close the issue, since it is now resolved for me.

closed 2020-04-24 by vslepakov