Azure/Industrial-IoT #393: Discovery show for Linux GW in 2.7.12 not for Windows

pietrobr, 2020-04-22:
I've installed the tool v 2.7.12 in discovery I see only the Linux Gateway, I expect to see the Windows as well.
[image]

dacolgit, 2020-04-23:
Please check in your IotHub if you have two edge devices: linuxgateway-xxx and wingat-xxx

[image]

pietrobr, 2020-04-23:
I've rebooted the simulation and GW Windows VM, now the discover works, but after I click on the  GW link for Win VM I don't see any endpoins
[image]
Twin modules are connected
[image]

cristipogacean, 2020-04-23:
@pietrobr: for windows you need change the scan mask like described here:
https://github.com/Azure/Industrial-IoT/blob/master/docs/tutorials/tut-discover-assets.md

the issue is that the default scan mask is generated for all the network adapters available in discovery container context. for windows only the internal docker network adapter is available therefore the discovery needs to be explicitly instructed to look for servers in the host's network adapter. Linux containers are smarter and have access to host's nw adapters as well.

closed 2020-04-27 by dacolgit