Azure/Industrial-IoT #395: Discovery result reported but not visible in the UI

vslepakov, 2020-04-24:
**Describe the bug**
discovery module was able to discover my opc-ua server and report the results back yo Azure but I cannot see any assets/endpoints in the engineering tool.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect an Ubuntu VM with IoT Edge to IoT Hub
2. Run [opc-plc](https://github.com/Azure-Samples/iot-edge-opc-plc) using
`docker run -d --name opc-plc -p 50000:50000 -v /home/<username>/opc-plc:/appdata 
 mcr.microsoft.com/iotedge/op
c-plc:1 -aa`
3. As soon as all modules are deployed, activate scanning in the engineering tool
4. Wait till you see in the logs of discovery module that it discovered the server and reported the result back to Azure

**Expected behavior**
I see my asset and the endpoint in the engineering tool

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browser: Chrome 
 - Version: Version: 80.0.3987.163 (Official Build) (64-bit)

**Additional context**
discovery module log:
```[10:52:07 WRN] Failed to add local host address. (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.DiscoveryServices)
System.Net.Internals.SocketExceptionFactory+ExtendedSocketException (00000005, 0xFFFDFFFF): Name or service not known
   at System.Net.Dns.InternalGetHostByName(String hostName)
   at System.Net.Dns.ResolveCallback(Object context)
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw(Exception source)
   at System.Net.Dns.HostResolutionEndHelper(IAsyncResult asyncResult)
   at System.Net.Dns.EndGetHostEntry(IAsyncResult asyncResult)
   at System.Net.Dns.<>c.<GetHostEntryAsync>b__27_1(IAsyncResult asyncResult)
   at System.Threading.Tasks.TaskFactory`1.FromAsyncCoreLogic(IAsyncResult iar, Func`2 endFunction, Action`1 endAction, Task`1 promise, Bool
ean requiresSynchronization)
--- End of stack trace from previous location where exception was thrown ---
   at Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.DiscoveryServices.AddLoopbackAddressesAsync(List`1 addresses) in D:\a\1\s\component
s\opc-ua\src\Microsoft.Azure.IIoT.OpcUa.Edge.Discovery\src\Services\DiscoveryServices.cs:line 512
[10:52:07 INF] : Starting port scanning (0 probes active)... (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher)
[10:52:07 INF] Probe 31 : found OPC UA server at [::ffff:172.17.0.4]:50000 (protocol:0) ... (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Servi
ces.DiscoveryServices)
[10:52:07 INF] : Found server 172.17.0.4:50000 (1656 scanned)... (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher)
[10:52:07 INF] Probe 0 : found OPC UA server at [::ffff:172.18.0.1]:50000 (protocol:0) ... (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Servic
es.DiscoveryServices)
[10:52:07 INF] : Found server 172.18.0.1:50000 (1656 scanned)... (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher)
[10:52:07 INF] Probe 4 : found OPC UA server at [::ffff:172.18.0.3]:50000 (protocol:0) ... (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Servic
es.DiscoveryServices)
[10:52:07 INF] : Found server 172.18.0.3:50000 (1656 scanned)... (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher)
[10:52:07 INF] Probe 375 : found OPC UA server at [::ffff:172.19.0.1]:50000 (protocol:0) ... (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Serv
ices.DiscoveryServices)
[10:52:07 INF] : Found server 172.19.0.1:50000 (1656 scanned)... (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher)
[10:52:07 INF] : Found 4 ports on servers (1656 scanned)... (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher)
[10:52:07 INF] : Searching 4 discovery urls for endpoints... (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher)
[10:52:07 INF] : Trying to find endpoints on opc.tcp://172.18.0.1:50000/... (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPubl
isher)
[10:52:08 INF] : Found 1 endpoints on opc.tcp://172.18.0.1:50000/. (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher)
[10:52:08 INF] : Trying to find endpoints on opc.tcp://industrial-edge.internal.cloudapp.net:50000/... (Microsoft.Azure.IIoT.OpcUa.Edge.Disc
overy.Services.ProgressPublisher)
[10:52:08 INF] : Found 1 endpoints on opc.tcp://industrial-edge.internal.cloudapp.net:50000/. (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Ser
vices.ProgressPublisher)
[10:52:08 INF] : Trying to find endpoints on opc.tcp://172.18.0.3:50000/... (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPubl
isher)
[10:52:08 INF] : Found 1 endpoints on opc.tcp://172.18.0.3:50000/. (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher)
[10:52:08 INF] : Trying to find endpoints on opc.tcp://172.19.0.1:50000/... (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPubl
isher)
[10:52:08 INF] : Found 1 endpoints on opc.tcp://172.19.0.1:50000/. (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher)
[10:52:08 INF] : Found total of 1 servers ... (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher)
[10:52:08 INF] Discovery took 00:00:25.4302853 and found 1 servers. (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.DiscoveryServices)
[10:52:08 INF] Uploading 1 results...(Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.DiscoveryServices)
[10:52:08 INF] 1 results uploaded. (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.DiscoveryServices)
[10:52:08 INF] : Discovery operation completed. (Microsoft.Azure.IIoT.OpcUa.Edge.Discovery.Services.ProgressPublisher)```

dacolgit, 2020-04-24:
when the scan is finished you should have something like this in the Eng tool:

[image]

vslepakov, 2020-04-24:
discovery is completed (module logs as above) but the "Scan Results" section is empty and there are no endpoints when I click on the discovery module name. happy to provide further logs if you need them.

vslepakov, 2020-04-29:
no idea why, but it just started working after a couple of days.

closed 2020-04-29 by vslepakov