Azure/Industrial-IoT #398: Postman tutorial: requests return html page of Engineering Tool with Authentication Failure

szmgabor, 2020-04-27:
**Describe the bug**
I was going through the steps described in the **Discover and register servers and browse their address space, read and publish nodes via REST API in Postman** tutorial from [here](https://github.com/Azure/Industrial-IoT/blob/master/docs/tutorials/tut-use-postman.md) and I stumbled upon a problem.
After successfully requesting an Access Token I was trying to run the first request, the **Registry: Get Applications**, but instead of receiving the applications I get in the response an HTML page, which is actually the starting page of the Engineering tool, with the Authentication Failer message.
I'm running version 2.7.11 of the Azure IIoT Platform.

**To Reproduce**
Steps to reproduce the behavior:
1. Follow the steps described in the Postman tutorial from [here](https://github.com/Azure/Industrial-IoT/blob/master/docs/tutorials/tut-use-postman.md)
2. After receiving the successfully obtaining an Access Token for the Azure Active Directory all the requests from the requests from the collection return in response an html, which is the landing page of the Engineering tool with the Authentication Failure message.

**Expected behavior**
I would expect the requests from the collection to work after the token was successfully obtained.

**Screenshots**
[image]
The name of the application is softing-edge-iiot, so the **OPC-SERVICEURL** is set to softing-edge-iiot.azurewebsites.net .
In the screenshot you can see that the request used the token.

**Desktop (please complete the following information):**
 - OS: Windows 10 Pro 1903 
 - Postman: Version 7.23.0

I might have missed something or not configured something properly, but I can't figure out what's the problem...

marcschier, 2020-04-28:
We need to update the documentation - The service is hosted in appservice under the application name if you deploy with -type services.  The service is under a different url when the engineering tool is also deployed in which case the engineering tool is under the application name and the service is under something like "servicexyzm".   The name of the service can be found in the portal in the resource group that was deployed.

szmgabor, 2020-04-29:
Thank you very much, it worked.
One remark though, the name in my case wasn't "servicexyzm" it was "svcxyzm", but I found it and it works now. Thanks again.

closed 2020-04-29 by szmgabor