Azure/Industrial-IoT #400: Azure IIoT Platform version switches during runtime

szmgabor, 2020-04-27:
**Describe the bug**
I've deployed on the 17th of April the Azure IIoT Platform using the deployment script, following the description from [here](https://github.com/Azure/Industrial-IoT/blob/master/docs/deploy/howto-deploy-all-in-one.md) It started up with version 2.7.11.
After a few days when I visited the page in the header it said 2.7.14 and also the layout of the navigation pane has changed.
And also till this date it's kind of random which version waits for me when I visit the page.

**To Reproduce**
Steps to reproduce the behavior:
1. Deployt the Azure IIot Platform according to the description frmo [here](https://github.com/Azure/Industrial-IoT/blob/master/docs/deploy/howto-deploy-all-in-one.md)
2. Visit the Engineering Tool home page of the application
3. The version number is sometimes 2.7.11 and sometimes it's 2.7.14

**Expected behavior**
I would expect that the version stays stable, if version 2.7.11 was deployed it should stay that way, or if there's a possibility to do on the fly updates I should get somehow notified.

**Screenshots**
With version 2.7.11:
[image]

With version 2.7.14:
[image]

Both these screenshot were taken today at a few F5 presses apart.

**Desktop (please complete the following information):**
 - OS: Windows 10 Pro Version 1903
 - Browser: Chrome
 - Version: 81.0.4044.122 (Official Build) (64-bit)

marcschier, 2020-04-28:
By default the deploy.ps1 deploys images with the "preview" tag.   These are floating and represent the latest state of master.   Thus when the app service restarts new images are pulled if they have changed.   It is possible to deploy a specific version using the -version argument of the deploy.ps1 script, e.g. "latest" or "2.7.11" in your case.   You can get the version of the master using the "get-version.ps1" script in tools/scripts.

marcschier, 2020-05-19:
To fixate a deployment to a version you can use the -version argument of the deploy.ps1 script and specify a particular version (as long as this version tag exists on mcr.microsoft.com).  E.g. ./deploy.ps1 -version 2.7.11 deploys this particular version.   

Caveat:  The services in 2.7.11 might require a slightly different set of services than are deployed by the arm templates in e.g. master branch if run from master.   E.g. -version 2.6.146 would require the script to be run from release/2.6.x branch to deploy the services as they are required for 2.6 release.

closed 2020-05-19 by marcschier