Azure/Industrial-IoT #437: More detailed OPC Publisher Module Tag documentation and better semantic

dbartz, 2020-05-09:
**Is your feature request related to a problem? Please describe.**
When deploying the OPC Publisher module, it is not trivial which docker image tag to choose:
* the documentation on [docker hub](https://hub.docker.com/_/microsoft-iotedge-opc-publisher) does not provide any information
* afais, tags without os/platform sometimes refer to windows images and sometimes to linux images
** `latest` refers to a linux image
** e.g.` 2.4.1`. refers to a windows image

**Describe the solution you'd like**
* the page on docker hub should feature more tags, in particular windows and linux containers
* tags should have a clear os/platform definition, there should be:
** latest tags with os/platform, e.g. `latest-windows-amd64`
** windows tags for higher versions, e.g. `2.7.7-windows-amd64` instead of just `2.7.7`

marcschier, 2020-05-09:
Hi @dbartz , all our images are multi arch, and in particular our module images run on all IoT Edge supported platforms.  However, if you have a scenario where you need a single platform image, please let us know.

dbartz, 2020-05-09:
Hi @marcschier,
let me describe how I stumbled over this issue.

I wanted to deploy the IoT Edge Gateway and the OPC Publisher on Windows without deploying the entire IoT Platform. To test the setup, I followed a mixture of two guides:
https://docs.microsoft.com/en-us/azure/iot-edge/quickstart
https://www.linkedin.com/pulse/step-by-step-guide-installing-opc-publisher-azure-iot-kevin-hilscher/

To summarize:
1. I create a windows VM
2. Deploy Iot-Edge
3. Connect IoT-Edge to an IoT Hub
4. Deploy OPC-Publisher Module via the IoT Hub Marketplace
    * this uses the OPC image with tag `latest`

The deployment fails. The Windows Service Logs claim that the reason is the wrong container. 
```
 276337 May 09 10:02  Warning     iotedged                        2 warn: edgelet_utils::logging --    caused by: image operating system "linux" cannot be used on this platform                                                                                              
  276336 May 09 10:02  Warning     iotedged                        2 warn: edgelet_utils::logging -- Could not pull image mcr.microsoft.com/iotedge/opc-publisher:latest
  276335 May 09 10:02  Information docker                          1 The description for Event ID '1' in Source 'docker' cannot be found.  The local computer may not have the necessary registry information or message DLL files to display the message, or you may not
                                                                     have permission to access them.  The following information is part of the event:'Attempting next endpoint for pull after error: image operating system "linux" cannot be used on this platform'
  276334 May 09 10:02  Information iotedged                        3 info: edgelet_docker::runtime -- Pulling image mcr.microsoft.com/iotedge/opc-publisher:latest...
```
I can solve this by choosing a different tag, e.g. `windows-amd64` or `2.7.9`

marcschier, 2020-05-15:
We validated this issue and found a "design issue" in our retagging script we run during release since docker apparently does not pull/push multiarch images.   

In the meantime, please use a dedicated release version such as "2.6.146" for the last release.

marcschier, 2020-05-19:
Hi, this should now be fixed.  I re-tagged latest as well as 2, and 2.6 as multi arch images.   Thanks for finding and reporting this issue:
  
```
docker run -it mplatform/mquery mcr.microsoft.com/iotedge/opc-publisher:latest
Image: mcr.microsoft.com/iotedge/opc-publisher:latest
 * Manifest List: Yes
 * Supported platforms:
   - windows/amd64:10.0.17763.973
   - linux/amd64
   - windows/amd64:10.0.18363.720
   - linux/arm64
   - linux/arm
```

closed 2020-05-19 by marcschier