Azure/Industrial-IoT #438: no logs in OPC Publisher 2.5.+

dbartz, 2020-05-10:
**Describe the bug**
When I use the OPC publisher module in version 2.4.1 (both linux and windows), I get log files in the folder I mount via the `bind` option (here the windows version):
```
"Binds": [
            "c:/iiotedge:c:/appdata"
        ],

```
When I use higher versions, no log files are written, both on Windows and Linux.  

Additional remarks:
* When I enter the Linux container via `docker exec`, I cannot find log files. 
* Changing the command line option `--lf` for the log file does not help

**To Reproduce**
Steps to reproduce the behavior:
1. [Deploy a virtual windows IoT Edge Device](https://docs.microsoft.com/en-us/azure/iot-edge/quickstart)
2. create folder `c:/iiotedge` on the vm 
3. add opc publisher module via marketplace
    * add the binding (see above)
    * select docker tag, e.g. `2.4.1` or `2.7.9`
4. For both versions, `iotedge list` reports the module as running
```PS C:\> iotedge list
NAME             STATUS           DESCRIPTION      CONFIG
OPCPublisher     running          Up a day         mcr.microsoft.com/iotedge/opc-publisher:2.7.9
edgeAgent        running          Up 3 days        mcr.microsoft.com/azureiotedge-agent:1.0
edgeHub          running          Up 3 days        mcr.microsoft.com/azureiotedge-hub:1.0
```
5. Depending on the container version, log files are written or not written:
    * `2.4.1` log files are written in `c:/iiotedge`
    * e.g. `2.7.9` no log files are written

**Expected behavior**
The OPC Publisher module logging should work 
* as described in the [documentation](https://docs.microsoft.com/en-us/azure/iot-accelerators/howto-opc-publisher-run)
* such that higher versions show backwards-compatible behaviour

I think the Issue is broader than just _no logging_, but without logging debugging is difficult.

cristipogacean, 2020-05-12:
The following IoT Edge module Container Create Options construct works fine for me on a Windows host system:
```
{
    "User": "ContainerAdministrator",
    "Cmd": [
        "--pf=/mount/publishednodes.json",
        "--tp=/mount/pki/trusted",
        "--rp=/mount/pki/rejected",
        "--ip=/mount/pki/issuer",
        "--lf=/mount/publisher.log",
        "--tm",
        "--fm=true",
        "--bs=500",
        "--si=20",
        "--di=1",
        "--sc=1"
    ],
    "NetworkingConfig": {
        "EndpointsConfig": {
            "host": {
                "IPAMConfig": {
                    "IPv4Address": "192.168.30.103"
                }
            }
        }
    },
    "HostConfig": {
        "NetworkMode": "host",
        "Mounts": [
            {
                "Type": "bind",
                "Source": "C:\\mount",
                "Target": "C:\\mount"
            }
        ]
    }
}
```

dbartz, 2020-05-13:
Hi @cristipogacean ,
thanks for the reply!
Which OPC Publisher Module image are you running? Could you e.g.  post the output of `iotedge list`?

cristipogacean, 2020-05-13:
I'm using the actual :preview which is 2.7.56

```
PS C:\Users\opcwall> iotedge list
NAME             STATUS           DESCRIPTION      CONFIG
discovery        running          Up 12 hours      mcr.microsoft.com/iotedge/discovery:2.7.56
edgeAgent        running          Up 12 hours      mcr.microsoft.com/azureiotedge-agent:1.0
edgeHub          running          Up 12 hours      mcr.microsoft.com/azureiotedge-hub:1.0
opcplc           running          Up 12 hours      mcr.microsoft.com/iotedge/opc-plc:latest
publisher        running          Up 12 hours      mcr.microsoft.com/iotedge/opc-publisher:2.7.56
twin             running          Up 12 hours      mcr.microsoft.com/iotedge/opc-twin:2.7.56
```

closed 2020-05-14 by cristipogacean

aagamb12, 2020-09-06:
Hey guys, 
I have a very similar issue,  I have succesfully deployed OPC publisher into my raspberry PI (already configured as IoTEdge device)  and everything seems to be properly configured in Azure OPC module container. When i run the command "iotedge list" you can see that the 3 services are running well but when i want to see the logs i could not get any logs in the OPC module, see image below. 
[image]

Here below is my my OPC container config file in Azure
[image]

my pn.json file located in the raspberry is below. 
[image]

i know there are OPC logs becuase i am running UA OPC client in my raspeberry which is collecting OPC tags from OPC server (located on the same VLAN) and i can see all tags coming into the OPC UA client running.  see image below
[image]

Any ideas what could be going on with the OPC publisher?  Thanks.
Thanks