Azure/Industrial-IoT #453: Latest master deployment succeeds but web apps fail with MSI issues

vslepakov, 2020-05-13:
**Describe the bug**
Web apps report an error when trying to access them

**To Reproduce**
Deploy latest master branch using deploy.ps1

**Expected behavior**
It works

**Logs from web app container**
`2020-05-13T18:26:02.234852326Z [18:26:02 DBG] Failed to authenticate to keyvault https://keyvault-ht5lez.vault.azure.net/ using Managed Service Identity: Parameters: Connection String: [No connection string specified], Resource: https://vault.azure.net, Authority: https://login.windows.net/72f988bf-86f1-41af-91ab-2d7cd011db47. Exception Message: Tried the following 3 methods to get an access token, but none of them worked.
2020-05-13T18:26:02.234882326Z Parameters: Connection String: [No connection string specified], Resource: https://vault.azure.net, Authority: https://login.windows.net/72f988bf-86f1-41af-91ab-2d7cd011db47. Exception Message: Tried to get token using Managed Service Identity. Access token could not be acquired. Received a non-retryable error. MSI ResponseCode: BadRequest, Response: {"statusCode":400,"message":"Unable to load requested managed identity.","correlationId":"83a50633-aea8-4c75-aefb-85719d46f824"}
2020-05-13T18:26:02.234890126Z Parameters: Connection String: [No connection string specified], Resource: https://vault.azure.net, Authority: https://login.windows.net/72f988bf-86f1-41af-91ab-2d7cd011db47. Exception Message: Tried to get token using Visual Studio. Access token could not be acquired. Environment variable LOCALAPPDATA not set.
2020-05-13T18:26:02.234895526Z Parameters: Connection String: [No connection string specified], Resource: https://vault.azure.net, Authority: https://login.windows.net/72f988bf-86f1-41af-91ab-2d7cd011db47. Exception Message: Tried to get token using Azure CLI. Access token could not be acquired. /bin/bash: az: No such file or directory`

vslepakov, 2020-05-13:
Switched to using **System Assigned Identity** as opposed to User Assigned identity which is created by the script and the web apps run now.

getting this in the **svc web app** now

`2020-05-13T19:46:15.629413115Z [19:46:15 FTL] Application startup exception (Microsoft.AspNetCore.Hosting.Diagnostics)
2020-05-13T19:46:15.629521516Z Autofac.Core.DependencyResolutionException: An exception was thrown while activating Microsoft.Azure.IIoT.Storage.Default.StartupMigration -> λ:Microsoft.Azure.IIoT.Storage.IMigrationTask[] -> Microsoft.Azure.IIoT.OpcUa.Registry.Migration.VaultApplicationMigration -> Microsoft.Azure.IIoT.Storage.CosmosDb.Services.CosmosDbServiceClient.
2020-05-13T19:46:15.629530716Z  ---> Autofac.Core.DependencyResolutionException: An exception was thrown while invoking the constructor 'Void .ctor(Microsoft.Azure.IIoT.Storage.CosmosDb.ICosmosDbConfig, Serilog.ILogger, Microsoft.Azure.IIoT.Storage.IJsonSerializerConfig)' on type 'CosmosDbServiceClient'.
2020-05-13T19:46:15.629536116Z  ---> System.ArgumentNullException: Value cannot be null. (Parameter 'DbConnectionString')
2020-05-13T19:46:15.629540916Z    at Microsoft.Azure.IIoT.Storage.CosmosDb.Services.CosmosDbServiceClient..ctor(ICosmosDbConfig config, ILogger logger, IJsonSerializerConfig jsonConfig) in D:\a\1\s\common\src\Microsoft.Azure.IIoT.Storage.CosmosDb\src\Services\CosmosDbServiceClient.cs:line 34
rc\Microsoft.Azure.IIoT.Services.All\src\Program.cs:line 26 ...`

And seeing this on the **non-svc web app**

[image]

marcschier, 2020-05-19:
Fixed in #463

closed 2020-05-19 by marcschier