Azure/Industrial-IoT #455: Could not send an Open Secure Channel request on Windows server 2019

alonsors-809, 2020-05-13:
Hi,

I am trying to connect the opc publisher with the opc server using certificates, but I am unable to. Both the server and the publisher are on the same Windows server 2019 machine. The folder pki/own contains the certificate that is auto created when the module runs and then I add the certificate of the server-client on the trusted folder. I set the Security Mode: true on the publishednodes_opcplc.json. I don’t know if I am missing a step installing certificates on the machine or something like that.

This is the Create Container Options of the module:

"Cmd": [
        "publisher",
        "--pf=/appdata/publishednodes_opcplc.json",
        "--di=60",
        "--to",
        "--aa",
        "--at=Directory",
        "--si=0",
        "--ms=0",
        "--ih=Amqp_Tcp_Only",
        "--tp=/appdata/pki/trusted",
        "--rp=/appdata/pki/rejected",
        "--ap=/appdata/pki/own",
        "--ip=/appdata/pki/issuer"
],

This are the security policies enabled:
 
[image]

Error log:
Session creation to endpoint 'opc.tcp://xxx.xxx.xx.xx:xxxx' failed 1 time(s). Please verify if server is up and Publisher configuration is correct.                                                                                                                           System.AggregateException: One or more errors occurred. (Error establishing a connection: Could not send an Open Secure Channel request.) ---> Opc.Ua.ServiceResultException: Error establishing a connection: Could not send an Open Secure Channel request.                                    at Opc.Ua.Client.Session.Create(ApplicationConfiguration configuration, ConfiguredEndpoint endpoint, Boolean updateBeforeConnect, Boolean checkDomain, String sessionName, UInt32 sessionTimeout, IUserIdentity identity, IList`1 preferredLocales)                                           --- End of inner exception stack trace ---                                                                                                     at System.Threading.Tasks.Task`1.GetResultCore(Boolean waitCompletionNotification)                                                             at OpcPublisher.OpcSession.ConnectSessionAsync(CancellationToken ct) in D:\a\1\s\modules\opc-publisher\opcpublisher\OpcSession.cs:line 444  ---> (Inner Exception #0) Opc.Ua.ServiceResultException: Error establishing a connection: Could not send an Open Secure Channel request.          at Opc.Ua.Client.Session.Create(ApplicationConfiguration configuration, ConfiguredEndpoint endpoint, Boolean updateBeforeConnect, Boolean checkDomain, String sessionName, UInt32 sessionTimeout, IUserIdentity identity, IList`1 preferredLocales)<---

cristipogacean, 2020-05-14:
We recently fixed the certificates recreation issue (#451). can you give it a try with the :preview tag container? 
Please be aware though that also the publisher's certificate must be trusted by the server.
Nevertheless, looking at the exception I cannot tell if this is a security issue or a connectivity issue. Can you please provide a bit more insight?

marcschier, 2020-05-19:
Closing - please could you try current master branch and re-open if the issue still exists.

closed 2020-05-19 by marcschier