Azure/Industrial-IoT #456: Deployment of a specific Release fails

vslepakov, 2020-05-14:
**Describe the bug**
From [Releases ](https://github.com/Azure/Industrial-IoT/releases) page I downloaded 2.6.146.zip so I could deploy this specific stable version. The deployment failed.

**To Reproduce**
Download version 2.6.146.zip from Releases, run deploy.ps1

**Expected behavior**
This specific version is deployed and works

**Additional context**
when running deploy.ps1 it outputs:
`Created new resource group vislepak-iiot-2-6-146 in northeurope.
WARNING: fatal: not a git repository (or any of the parent directories): .git.  Using master branch.
Deployment will use 'master' branch in 'https://github.com/Azure/Industrial-IoT'.`

I do not want to deploy the master branch though...or does it use a specific commit/tag?

then it fails with:
`Error: Code=InvalidContentLink; Message=Unable to download deployment content from 'https://raw.githubusercontent.com/Azure/Industrial-IoT/master/deploy/templates/azuredeploy.infrastructure.json'. The tracking Id is '35c1750a-df02-4bd7-a7e2-75b3975c067a'. Please see https://aka.ms/arm-deploy for usage details.`

marcschier, 2020-05-14:
We need to fail running the deploy.ps1 if run outside of Git, as we cannot control the content of the release.zip.   We will support deploy to azure eventually.

vslepakov, 2020-05-14:
Ok, what is the recommended way for now do deploy a specific release? master is not always stable.

marcschier, 2020-05-15:
Hi @vslepakov, You can use the IAI deployment tool or the deploy.ps1 in release/2.6.x branch (last release, deploys "latest").   You can also use a specific version tag or "latest" tag as input into the deploy.ps1, and it will use these images instead of "preview".

vslepakov, 2020-05-15:
Got it, thanks!

closed 2020-05-15 by vslepakov