Azure/Industrial-IoT #464: Deploy OPC Twin and OPC Discovery in Standalone mode

ma3goul, 2020-05-18:
**Is your feature request related to a problem? Please describe.**
Hi All ,

I prototyped a simple POC using a simulation OPC UA server and OPC publisher with legacy configuration .
Now I want to run an OPC twin with the OPC publisher .

**Describe the solution you'd like**
Is it possible to run OPC twin (and maybe OPC discovery) in standalone mode ? without deploying the whole industrial iot platform ?

**Describe alternatives you've considered**
Is it possible also to call methods via the OPC TWIN using a customised module running on the edge ?

Thank You in advance .

marcschier, 2020-05-18:
Hi,. Thank you for the request. It is not possible at this point.  We will however add it to our internal long term planning backlog.

PWSys, 2020-05-18:
This would be helpful!

rowenwu, 2020-06-24:
Hi @ma3goul @PWSys, I'm a PM helping to coordinate the internal planning. Would you like to set up some time to chat so that we can capture your requirements and align it with our backlog? If so, feel free to email me at rowwu@microsoft.com.

Adrian-Wennberg, 2022-11-14:
Has there been any progress on this? I am working on a project where we want to add the Twin module and microservice to our existing setup. We are currently subsribing to OPC Publisher though an IoT edge device, and we need to add on the ability to set values on the OPC server. 

As far as we can tell, the twin module and microservice is our best bet, and we want to add those two, and only those two if we can to our existing deployment. Is there a way to do this at the moment?

Adrian-Wennberg, 2023-05-16:
I assume yes you can do it with a C# module. I ended up getting in contact with the IoT team directly and they gave me the pointers I needed to set up a javascript module module using the node-opcua package. The biggest hurdle for me was learning to develop and deploy my own module, but it was otherwise fairly simple 🙂

PWSys, 2023-05-16:
This is possible using a custom module.  

Here is a basic write example using the OPC-UA .Net Standard library:
https://github.com/OPCFoundation/UA-.NETStandard/blob/29a3be5533ce4027f394995a7343c1ce91969851/Applications/ConsoleReferenceClient/ClientSamples.cs#L141

marcschier, 2023-05-19:
We released 2.9.0-preview2 which has these capabilities, documentation regarding API and samples are following in the next months.

closed 2023-05-19 by marcschier